snaptec/openWB #1878: Discovergy broken durch #1802 (?)

cstueckrath, 2022-01-05:
Es scheint da ein Problem zu geben mit dem Update gestern:

`KeyError: 'voltage1'
    voltages = [values["voltage" + str(phase)] / 1000 for phase in range(1, 4)]
  File "/var/www/html/openWB/modules/bezug_discovergy/discovergy.py", line 27, in <listcomp>
    voltages = [values["voltage" + str(phase)] / 1000 for phase in range(1, 4)]
  File "/var/www/html/openWB/modules/bezug_discovergy/discovergy.py", line 27, in write_readings_to_ramdisk
    write_readings_to_ramdisk(get_last_reading(user, password, meter_id))
  File "/var/www/html/openWB/modules/bezug_discovergy/discovergy.py", line 42, in update
    parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
    args.RUN(args)
  File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 34, in run_using_positional_cli_args
    run_using_positional_cli_args(update)
  File "/var/www/html/openWB/modules/bezug_discovergy/discovergy.py", line 47, in <module>
t):`

yankee42, 2022-01-05:
Bitte verwende Dreifach-Backticks statt einfache Backticks um die Fehlermeldung zu markieren:

    ```Hier zwischen 3x Backticks die Fehlermeldung```

So lässt sich der Text kaum lesen.

Könntest du einmal die URL https://api.discovergy.com/public/v1/last_reading?meterId=XXXX aufrufen (XXXX ersetzen durch deine Meter-ID, wie du sie auch in der Modulkonfiguration abgelegt hast) (du solltest nach Benutzername und Passwort gefragt werden, ebenfalls so angeben wie in der Modulkonfig) und das Ergebnis hier rein posten?

LKuemmel, 2022-01-05:
PR #1802 hat den Fix aus PR #1514 entfernt.

cstueckrath, 2022-01-05:
sorry, Formatierung gefixt ;)

```{"time":1641374253118,"values":{"phase2Power":241970,"energyOut":7690248872000,"energy2":28550717775000,"energy1":2258421000,"phase1Power":81310,"power":485890,"phase3Power":162600,"phase1Voltage":225500,"energy":28552976196000,"phase2Voltage":226400,"phase3Voltage":226800}}```

LKuemmel, 2022-01-05:
Jetzt ist die Antwort von deinem Zähler weg ;) Kannst Du sie bitte nochmal posten, dann würde ich sie in den Test für das Discovergy-Modul einbauen?

yankee42, 2022-01-05:
> Jetzt ist die Antwort von deinem Zähler weg ;) Kannst Du sie bitte nochmal posten, dann würde ich sie in den Test für das Discovergy-Modul einbauen?

Die ist schon noch da?!

Jedenfalls interessant, dass du  @cstueckrath ein anderes JSON bekommst als ich auf meinem Meter. Es gibt da offensichtlich unterschiedliche Formate.

@LKuemmel: Deine Antwort interpretiere ich so, dass du dich darum kümmerst?

cstueckrath, 2022-01-05:
hinter values: ist eigentlich ein Umbruch. Der wird hier nur nicht angezeigt, warum auch immer...

LKuemmel, 2022-01-05:
Oh, Neuladen der Seite hat geholfen. Davor war das Feld einfach leer.
@yankee42 : Ja, mache ich.

cstueckrath, 2022-01-05:
ich habe einen Meteroit V3 (der grüne Zähler). Ist deiner @yankee42 vielleicht ein anderer?

LKuemmel, 2022-01-05:
Bitte mal ausprobieren, ob das Problem mit PR #1881 behoben ist.

cstueckrath, 2022-01-05:
Läuft!

closed 2022-01-05 by cstueckrath

yankee42, 2022-01-05:
> ich habe einen Meteroit V3 (der grüne Zähler). Ist deiner @yankee42 vielleicht ein anderer?

Jo, ich habe noch einen V2.