snaptec/openWB #1956: Steuerung im "Nur Ladepunkt" Modus funktioniert nicht wie erwartet

andig, 2022-01-22:
Wir wollen die owb im "Nur Ladepunkt" Modus steuern. Die verwendeten Topics entsprechen https://github.com/snaptec/openWB/blob/10bced471857f1c2af40e7b37dc8ba156d951746/runs/mqttsub.py#L1218.

Die Steuerung funktioniert prinzipiell, d.h. wir können feststellen dass das Auto verbunden ist/ lädt. Aber:

- [ ] der eingestellte Strom (`ADirectChargeAmps`) unter `openWB/lp1` oder `openWB/lp/1` wird nicht aktualisiert (das ist laut @snaptec erwartet)
- [ ] die Ströme `APhase1...3` unter dem Ladepunkt werden nicht aktualisiert
- [ ] die Energie `kWhCounter` unter dem Ladepunkt werden nicht aktualisiert
- [ ] die OpenWB schickt keine oder nur verzögerte Updates mehr über `openWB/system/Timestamp` obwohl `openWB/set/isss/heartbeat` laufend mit `1` beschrieben wird

      [openwb] TRACE 2022/01/22 12:14:42 recv openWB/system/Timestamp: '1642850072' <- letztes Update (müsste alle 10s kommen)
      [openwb] TRACE 2022/01/22 12:14:52 send openWB/set/isss/heartbeat: '1'        <- hb wird aktualisiert
      [openwb] ERROR 2022/01/22 12:14:54 heartbeat: timeout                         <- mehr als 10s vergangen, kein neuer Timestamp erhalten
      [openwb] TRACE 2022/01/22 12:15:02 send openWB/set/isss/heartbeat: '1'
      [openwb] ERROR 2022/01/22 12:15:04 heartbeat: timeout
      [openwb] TRACE 2022/01/22 12:15:12 send openWB/set/isss/heartbeat: '1'
      [openwb] ERROR 2022/01/22 12:15:14 heartbeat: timeout
      [openwb] TRACE 2022/01/22 12:15:22 send openWB/set/isss/heartbeat: '1'
      [openwb] TRACE 2022/01/22 12:15:24 recv openWB/system/Timestamp: '1642850092'
      [openwb] TRACE 2022/01/22 12:15:24 recv openWB/system/Timestamp: '1642850102'
      [openwb] TRACE 2022/01/22 12:15:24 recv openWB/system/Timestamp: '1642850112'
      [openwb] TRACE 2022/01/22 12:15:24 recv openWB/system/Timestamp: '1642850122'
      [openwb] TRACE 2022/01/22 12:15:32 recv openWB/system/Timestamp: '1642850132'

/cc @snaptec

snaptec, 2022-01-22:
Was sagt unter Status das nur Ladepunkt log der openWB?

snaptec, 2022-01-22:
Status url kopieren, Nur Ladepunktmodus aktivieren, einmal neustarten. 

Nun beobachten. 

Laut Log klappt hier eher die zählerauslesung generell nicht. 

On top dann bitte gleich einen Systembericht der openWB senden (Einstellungen, System, Support) und das Issue hier verlinken.

andig, 2022-01-23:
Wir haben jetzt nochmal auf Nighly aktualisiert. Keine Fehler mehr im `isss.log` (Evtl. aufgrund https://github.com/snaptec/openWB/commit/64fbaa2b69b34c5c51113c13459e845f26758889?), aber dennoch keine Updates auf MQTT. Wir prüfen das jetzt nochmal durch, Systembericht folgt.

andig, 2022-01-23:
> Status url kopieren, Nur Ladepunktmodus aktivieren, einmal neustarten.

Neustart nur über Shell möglich da Menüpunkt nicht mehr erreichbar.

Jetzt Auto (i3) verbunden. `openWB/lp/1/boolPlugStat` bleibt auf `0`. Ladekabel verriegelt nicht. Nur Ladepunkt Log ist leer: 

- [image]
- [image]

Meine These: OpenWB hängt irgendwo, zumindest "vergisst" es, auch MQTT zu aktualisieren. Kann es sein, dass die Updates nur auf eine Master OpenWB passieren anstatt auf den Ladepunkt? Das wäre natürlich sinnlos ;)

Systembericht senden aus Nur Ladepunkt Modus ist nicht möglich da Datenschutzbestimmung nicht akzeptiert werden kann und immer wieder Redirect auf die Nur Ladepunkt Oberfläche kommt (nerv...).

Nach rückstellen auf Normalbetriebsmodus wird plötzlich auch `openWB/lp/1/boolPlugStat` wieder aktualisiert.
Ticket ID: #67052275

andig, 2022-01-23:
Auf MQTT ist folgendes los im Moment des "Ladestarts" (der nicht passiert):

    mosquitto_sub -t openWB/set/# -v -h <ip>
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/Current 16
    openWB/set/isss/Current 16
    openWB/set/isss/Current (null)
    openWB/set/isss/Current (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1
    openWB/set/isss/heartbeat (null)
    openWB/set/isss/heartbeat 1

Lp1 stellt sich vollkommen tot:

    mosquitto_sub -t openWB/lp/1/# -v -h <ip>

andig, 2022-01-30:
Ich hab eine Ahnung. Kann es sein, dass die OpenWB die Nachrichten nur sieht, wenn sie als `retained` geschickt werden? Das würde es erklären...

Wie funktioniert der Empfangsprozess? Lauscht da etwas ständig oder nur alle 10s?

snaptec, 2022-02-18:
geklärt per Slack.

closed 2022-02-18 by snaptec

andig, 2022-02-18:
Lösung: MQTT messages *müssen* `retained` sein.