snaptec/openWB #1964: Parsing von SMA-HomeManager nach Update Fehlerhaft

atwebm, 2022-01-26:
Wer die "Direkte Zähler Kommunikation" (=Unicast) eingestellt hat, um evtl. IGMP-Problematiken im eigenen Netz zu umgehen, der erhält seit dem Update vom SHM 2.0 Datagramme mit der Länge von 608 Bytes (=identischer Aufbau zu den Multicast-Datagrammen) anstelle von wie bislang 610 Byte. Das Parsing muss also angepasst werden bzw. erfordert es keine Unterscheidung mehr zwischen Multicast- und Unicast-Datagrammen.

Wäre natürlich schick gewesen von SMA, das auch kundzutun,

Ich hab das ausprobiert und seit dem leeren von den Wechselrichter-IPs (in der Sunnyportal-Konfiguration) werden wieder Werte in der OpenWB angezeigt.

Vermuteter Source:
https://github.com/snaptec/openWB/blob/master/modules/bezug_smashm/speedwiredecoder.py

yankee42, 2022-01-26:
Ist das mit der aktuellen Nightly verifiziert? Mit #1845 wurde schon etwas verändert.

Mir ist von dem Unicast+SMA überhaupt nichts bekannt. Wie funktioniert das dann?

atwebm, 2022-01-26:
installierte Version: 1.9.252 (2022-01-25 15:32:04 +0100 [62fad1f0] )

Hier hat das jemand zusammen geschrieben: https://www.photovoltaikforum.com/thread/118604-st%C3%B6rung-der-kommunikation-zur-z%C3%A4hlereinrichtung-und-fw/?postID=2121551#post2121551

Man kann da direkt die IP-Adressen der Wechselrichter eintragen und dann wird das zusaetzlich per Unicast dorthin geschickt.
Somit umgeht man IGMP-Probleme.

yankee42, 2022-01-26:
OK, verstehe.

Könntest du mit Hilfe von dem Code aus #1853 Testdaten ermitteln?

atwebm, 2022-01-26:
Zuerst mit eingetragenen IPs (direkte Kommunikation).
Dannach im Sunnyportal die IPs geloescht und ca. eine Minute spaeter dann wieder richtig.

root@openWB:/var/www/html/openWB/ramdisk# python sma-evu-test.py 
-----
U01BAAAEAqAAAAABACYAEGBlCaD///////8AAHQBd+RWswAAAAAAAD6hAAIAAAAAAAAAAAAAAAAAA
A==
.....
-----
U01BAAAEAqAAAAABACYAEGBlCaD///////8AAHQBd+RWswAAAAAAALKhAAIAAAAAAAAAAAAAAAAAA
A==
.....
-----
U01BAAAEAqAAAAABACYAEGBlCaD///////8AAHQBd+RWswAAAAAAAB+iAAIAAAAAAAAAAAAAAAAAA
A==
.....
-----
U01BAAAEAqAAAAABACYAEGBlCaD///////8AAHQBd+RWswAAAAAAAIeiAAIAAAAAAAAAAAAAAAAAA
A==
.....
-----
U01BAAAEAqAAAAABAkwAEGBpAXSzVuR3lkNZaAABBAAAAAAAAAEIAAAAAABeWaroAAIEAAAAwtUAA
ggAAAAACQy6JEgAAwQAAAAAAAADCAAAAAAADW12eAAEBAAAAAYnAAQIAAAAAAMMFaaAAAkEAAAAAA
AACQgAAAAAAbSY9XAACgQAAADC7gAKCAAAAAAKS3RxEAANBAAAAAPoAA4EAAAAwyEAFQQAAAAAAAA
VCAAAAAAA6VUDGAAWBAAAAH3pABYIAAAAAAWmZ7cIABcEAAAAAAAAFwgAAAAAABNUlTAAGAQAAAAC
JgAYCAAAAAABJXLFOAAdBAAAAAAAAB0IAAAAAAEHxMH4AB4EAAAAfe0AHggAAAAABgOEM6gAHwQAA
AA2ZgAgBAAAA4ooACEEAAAAA+gAKQQAAAAAAAApCAAAAAAA0ZoDGAAqBAAAACGXACoIAAAAAAJpvL
eYACsEAAAAAAAAKwgAAAAAAAFZbTgALAQAAAADhwAsCAAAAAABXIF/sAAxBAAAAAAAADEIAAAAAAF
maQGIADIEAAAAIcYAMggAAAAAAnonFMAAMwQAAAAO2gA0BAAAA32tADUEAAAAA+MAPQQAAAAAAAA9
CAAAAAAARxjrAAA+BAAAACNVAD4IAAAAAAKgRAXIAD8EAAAAAAAAPwgAAAAAAA4oZ/gAQAQAAAAAe
wBACAAAAAAAn4pRSABFBAAAAAAAAEUIAAAAAACpyFpgAEYEAAAAI1YARggAAAAAAqQ9ccgARwQAAA
APdwBIBAAAA4GXAEkEAAAAA+iQAAAAAggFUgAAAAA=
.....
-----
U01BAAAEAqAAAAABAkwAEGBpAXSzVuR3lkNdVQABBAAAAAAAAAEIAAAAAABeWaroAAIEAAAAwqQAA
ggAAAAACQy6NpAAAwQAAAAAAAADCAAAAAAADW12eAAEBAAAAAXjAAQIAAAAAAMMFaaAAAkEAAAAAA
AACQgAAAAAAbSY9XAACgQAAADCuwAKCAAAAAAKS3SEwAANBAAAAAPoAA4EAAAAwx0AFQQAAAAAAAA
VCAAAAAAA6VUDGAAWBAAAAH3QABYIAAAAAAWmZ8OwABcEAAAAAAAAFwgAAAAAABNUlTAAGAQAAAAC
RQAYCAAAAAABJXLFOAAdBAAAAAAAAB0IAAAAAAEHxMH4AB4EAAAAfdUAHggAAAAABgOEQFAAHwQAA
AA2ZAAgBAAAA4mwACEEAAAAA+gAKQQAAAAAAAApCAAAAAAA0ZoDGAAqBAAAACFkACoIAAAAAAJpvL
poACsEAAAAAAAAKwgAAAAAAAFZbTgALAQAAAADhgAsCAAAAAABXIGBGAAxBAAAAAAAADEIAAAAAAF
maQGIADIEAAAAIZQAMggAAAAAAnonGPgAMwQAAAAOxwA0BAAAA309ADUEAAAAA+IAPQQAAAAAAAA9
CAAAAAAARxjrAAA+BAAAACNwAD4IAAAAAAKgRAiYAD8EAAAAAAAAPwgAAAAAAA4oZ/gAQAQAAAAAG
QBACAAAAAAAn4pRSABFBAAAAAAAAEUIAAAAAACpyFpgAEYEAAAAI3AARggAAAAAAqQ9dJgARwQAAA
APjwBIBAAAA4EIAEkEAAAAA+iQAAAAAggFUgAAAAA=
.....
-----
U01BAAAEAqAAAAABAkwAEGBpAXSzVuR3lkNhOAABBAAAAAAAAAEIAAAAAABeWaroAAIEAAAAwm4AA
ggAAAAACQy6SkAAAwQAAAAAAAADCAAAAAAADW12eAAEBAAAAAZFAAQIAAAAAAMMFaaAAAkEAAAAAA
AACQgAAAAAAbSY9XAACgQAAADCiAAKCAAAAAAKS3SYcAANBAAAAAPnAA4EAAAAwxEAFQQAAAAAAAA
VCAAAAAAA6VUDGAAWBAAAAH4SABYIAAAAAAWmZ9BYABcEAAAAAAAAFwgAAAAAABNUlTAAGAQAAAAC
NQAYCAAAAAABJXLGoAAdBAAAAAAAAB0IAAAAAAEHxMH4AB4EAAAAfhcAHggAAAAABgOETPgAHwQAA
AA2cAAgBAAAA4qpACEEAAAAA+gAKQQAAAAAAAApCAAAAAAA0ZoDGAAqBAAAACF0ACoIAAAAAAJpvL
04ACsEAAAAAAAAKwgAAAAAAAFZbTgALAQAAAADngAsCAAAAAABXIGBGAAxBAAAAAAAADEIAAAAAAF
maQGIADIEAAAAIaYAMggAAAAAAnonG8gAMwQAAAAOygA0BAAAA35HADUEAAAAA+IAPQQAAAAAAAA9
CAAAAAAARxjrAAA+BAAAACLoAD4IAAAAAAKgRAzQAD8EAAAAAAAAPwgAAAAAAA4oZ/gAQAQAAAAAc
gBACAAAAAAAn4pRSABFBAAAAAAAAEUIAAAAAACpyFpgAEYEAAAAIugARggAAAAAAqQ9eNAARwQAAA
APVQBIBAAAA39oAEkEAAAAA+iQAAAAAggFUgAAAAA=
.....
-----
U01BAAAEAqAAAAABAkwAEGBpAXSzVuR3lkNlJQABBAAAAAAAAAEIAAAAAABeWaroAAIEAAAAwskAA
ggAAAAACQy6XfAAAwQAAAAAAAADCAAAAAAADW12eAAEBAAAAAZVAAQIAAAAAAMMFafoAAkEAAAAAA
AACQgAAAAAAbSY9XAACgQAAADC4wAKCAAAAAAKS3SsIAANBAAAAAPnAA4EAAAAwx8AFQQAAAAAAAA
VCAAAAAAA6VUDGAAWBAAAAH4FABYIAAAAAAWmZ90AABcEAAAAAAAAFwgAAAAAABNUlTAAGAQAAAAC
LQAYCAAAAAABJXLGoAAdBAAAAAAAAB0IAAAAAAEHxMH4AB4EAAAAfgoAHggAAAAABgOEWaAAHwQAA
AA2bQAgBAAAA4qHACEEAAAAA+gAKQQAAAAAAAApCAAAAAAA0ZoDGAAqBAAAACGDACoIAAAAAAJpvM
FwACsEAAAAAAAAKwgAAAAAAAFZbTgALAQAAAADowAsCAAAAAABXIGBGAAxBAAAAAAAADEIAAAAAAF
maQGIADIEAAAAIbYAMggAAAAAAnonHpgAMwQAAAAO0gA0BAAAA34iADUEAAAAA+IAPQQAAAAAAAA9
CAAAAAAARxjrAAA+BAAAACNAAD4IAAAAAAKgRA+gAD8EAAAAAAAAPwgAAAAAAA4oZ/gAQAQAAAAAh
QBACAAAAAAAn4pRSABFBAAAAAAAAEUIAAAAAACpyFpgAEYEAAAAI0EARggAAAAAAqQ9e6AARwQAAA
[... 73 lines cut ...]
AA2ZgAgBAAAA4qeACEEAAAAA+gAKQQAAAAAAAApCAAAAAAA0ZoDGAAqBAAAACGAACoIAAAAAAJpvN
UgACsEAAAAAAAAKwgAAAAAAAFZbTgALAQAAAADjgAsCAAAAAABXIGD6AAxBAAAAAAAADEIAAAAAAF
maQGIADIEAAAAIbAAMggAAAAAAnonM7AAMwQAAAAO0QA0BAAAA33UADUEAAAAA+IAPQQAAAAAAAA9
CAAAAAAARxjrAAA+BAAAACKIAD4IAAAAAAKgRCS4AD8EAAAAAAAAPwgAAAAAAA4oZ/gAQAQAAAAAk
gBACAAAAAAAn4pRSABFBAAAAAAAAEUIAAAAAACpyFpgAEYEAAAAIokARggAAAAAAqQ9kLgARwQAAA
APIwBIBAAAA4HQAEkEAAAAA+iQAAAAAggFUgAAAAA=
.....
-----
U01BAAAEAqAAAAABAkwAEGBpAXSzVuR3lkOAeAABBAAAAAAAAAEIAAAAAABeWaroAAIEAAAAwjIAA
ggAAAAACQy65lgAAwQAAAAAAAADCAAAAAAADW12eAAEBAAAAAYkAAQIAAAAAAMMFawgAAkEAAAAAA
AACQgAAAAAAbSY9XAACgQAAADCSwAKCAAAAAAKS3UzIAANBAAAAAPoAA4EAAAAwxUAFQQAAAAAAAA
VCAAAAAAA6VUDGAAWBAAAAH3yABYIAAAAAAWmaDQwABcEAAAAAAAAFwgAAAAAABNUlTAAGAQAAAAC
GgAYCAAAAAABJXLICAAdBAAAAAAAAB0IAAAAAAEHxMH4AB4EAAAAffYAHggAAAAABgOEsNAAHwQAA
AA2ZAAgBAAAA4qPACEEAAAAA+gAKQQAAAAAAAApCAAAAAAA0ZoDGAAqBAAAACEDACoIAAAAAAJpvN
lYACsEAAAAAAAAKwgAAAAAAAFZbTgALAQAAAADiwAsCAAAAAABXIGD6AAxBAAAAAAAADEIAAAAAAF
maQGIADIEAAAAITQAMggAAAAAAnonNoAAMwQAAAAOmwA0BAAAA321ADUEAAAAA+IAPQQAAAAAAAA9
CAAAAAAARxjrAAA+BAAAACM9AD4IAAAAAAKgRCjwAD8EAAAAAAAAPwgAAAAAAA4oZ/gAQAQAAAAAf
wBACAAAAAAAn4pRSABFBAAAAAAAAEUIAAAAAACpyFpgAEYEAAAAIz4ARggAAAAAAqQ9lPAARwQAAA
APawBIBAAAA4HTAEkEAAAAA+iQAAAAAggFUgAAAAA=
.....
^CTraceback (most recent call last):
  File "sma-evu-test.py", line 22, in <module>
    buffer = sock.recv(1024)
KeyboardInterrupt
root@openWB:/var/www/html/openWB/ramdisk#

atwebm, 2022-01-26:
Forum-Link: https://openwb.de/forum/viewtopic.php?p=56357#p56357

yankee42, 2022-02-08:
Die von dir geposteten Datenpakete lassen sich parsen. Im Forenthread steht erledigt. Ist das Ticket damit auch erledigt?

atwebm, 2022-02-13:
Ja. Ich wollte damit nur auf das Problem hinweisen.
Ich konnte wieder auf Multicast zurueck-Schalten und damit das Problem umgehen.

closed 2022-02-13 by atwebm