snaptec/openWB #1968: MQTT EVU: Hausverbrauch nur als Ganzzahl

Alligator09, 2022-01-26:
openWB Version 1.9.244 

Beim Setzen des MQTT Topics "openWB/set/evu/W" ist mir aufgefallen, dass openWB bei Werten mit Nachkommastelle nicht richtig rechnet und stattdessen eine 0 annimmt. Wenn das Smartmeter (hier SDM 630) mal zufällig eine ganzzahlige Leistung misst, verarbeitet openWB den Wert korrekt.

Als Workaround formatiere ich die Leistung in openHAB so, dass sie keine Nachkommastellen hat, dann werden sie von openWB zuverlässig verarbeitet. (s. Diagramm unten: zunächst nur Spitzen, wenn zufällig eine Ganzzahl, am Ende, nach Abschneiden der Nachkommastellen, eine durchgängige Linie.)

Ist es gewünscht, dass openWB auf dem Topic keine Nachkommastellen verarbeitet, oder ist es ein Bug?

[image]

benderl, 2022-01-27:
Das ist absichtlich so und auch in der Modulkonfiguration so beschrieben.

Alligator09, 2022-01-27:
Danke für die schnelle Antwort. Den Datentyp habe ich offensichtlich überlesen.

closed 2022-01-27 by Alligator09