snaptec/openWB #1980: SMA EM/SHM Modul erfordert Seriennummer, Parsing der empfangenen Daten klappt nicht.

MartinRinas, 2022-02-01:
@LKuemmel ich hab' eben ein Update auf die aktuelle Nightly mit dem #1979 gemacht, leider werden die Daten von meinem SMA EM nun nicht  mehr verarbeitet.

Scheint hier zwei Fehlersituationen zu geben. 

1) Zum einen schlägt der Aufruf des legacy run server fehl wenn keine Seriennummer in der Modulkonfiguration hinterlegt ist. Bisher war diese optional und nur notwending sofern mehrere EM/SHMs im Netzwerk vorhanden sind.
`legacy_run_server.py: error: argument serials: invalid int value: 'none'
2022-02-01 09:05:43: PID: 4743: legacy run server: usage: legacy_run_server.py [-h] component_type serials [num]`

2) Seriennummer eingtragen, das Datagram kommt lt. Log an, wird dann aber nicht korrekt weiterverarbeitet, zumindest bleiben alle EVU Daten bei mir auf 0.
`2022-02-01 09:51:34: EVU RET: 0 (LV2) at 24 main modules/bezug_smashm/main.sh
2022-02-01 09:51:34: PID: 4743: legacy run server: Completed running command in 0.86s: ["modules.sma.device","counter","1900210416"]
2022-02-01 09:51:34: PID: 4743: root: Update completed successfully
2022-02-01 09:51:34: PID: 4743: root: SMA-Datagramm: {'serial': 1900210416, 'pconsume': 1206.7, 'pconsumeunit': 'W', 'pconsumecounter': 30697.4513, 'pconsumecounterunit': 'kWh', 'psupply': 0.0, 'psupplyunit': 'W', 'psupplycounter': 71579.4217, 'psupplycounterunit': 'kWh', 'sconsume': 0.0, 'sconsumeunit': 'VA', 'sconsumecounter': 9489.0698, 'sconsumecounterunit': 'kVAh', 'ssupply': 282.4, 'ssupplyunit': 'VA', 'ssupplycounter': 10789.6408, 'ssupplycounterunit': 'kVAh', 'qconsume': 1239.3, 'qconsumeunit': 'VAr', 'qconsumecounter': 33181.0522, 'qconsumecounterunit': 'kVArh', 'qsupply': 0.0, 'qsupplyunit': 'VAr', 'qsupplycounter': 73458.2289, 'qsupplycounterunit': 'kVArh', 'cosphi': 0.974, 'cosphiunit': '°', 'p1consume': 0.0, 'p1consumeunit': 'W', 'p1consumecounter': 14249.3564, 'p1consumecounterunit': 'kWh', 'p1supply': 74.9, 'p1supplyunit': 'W', 'p1supplycounter': 25442.0749, 'p1supplycounterunit': 'kWh', 's1consume': 0.0, 's1consumeunit': 'VA', 's1consumecounter': 3677.5071, 's1consumecounterunit': 'kVAh', 's1supply': 34.4, 's1supplyunit': 'VA', 's1supplycounter': 3874.8395, 's1supplycounterunit': 'kVAh', 'q1consume': 0.0, 'q1consumeunit': 'VAr', 'q1consumecounter': 15288.1176, 'q1consumecounterunit': 'kVArh', 'q1supply': 82.4, 'q1supplyunit': 'VAr', 'q1supplycounter': 26173.9742, 'q1supplycounterunit': 'kVArh', 'i1': 0.43, 'i1unit': 'A', 'u1': 227.913, 'u1unit': 'V', 'cosphi1': 0.908, 'cosphi1unit': '°', 'p2consume': 976.0, 'p2consumeunit': 'W', 'p2consumecounter': 8751.5933, 'p2consumecounterunit': 'kWh', 'p2supply': 0.0, 'p2supplyunit': 'W', 'p2supplycounter': 24948.9095, 'p2supplycounterunit': 'kWh', 's2consume': 0.0, 's2consumeunit': 'VA', 's2consumecounter': 3724.6421, 's2consumecounterunit': 'kVAh', 's2supply': 202.6, 's2supplyunit': 'VA', 's2supplycounter': 2898.5446, 's2supplycounterunit': 'kVAh', 'q2consume': 996.8, 'q2consumeunit': 'VAr', 'q2consumecounter': 9776.8341, 'q2consumecounterunit': 'kVArh', 'q2supply': 0.0, 'q2supplyunit': 'VAr', 'q2supplycounter': 25590.7989, 'q2supplycounterunit': 'kVArh', 'i2': 4.479, 'i2unit': 'A', 'u2': 225.167, 'u2unit': 'V', 'cosphi2': 0.979, 'cosphi2unit': '°', 'p3consume': 305.6, 'p3consumeunit': 'W', 'p3consumecounter': 10739.042, 'p3consumecounterunit': 'kWh', 'p3supply': 0.0, 'p3supplyunit': 'W', 'p3supplycounter': 24230.9778, 'p3supplycounterunit': 'kWh', 's3consume': 0.0, 's3consumeunit': 'VA', 's3consumecounter': 2975.1725, 's3consumecounterunit': 'kVAh', 's3supply': 45.4, 's3supplyunit': 'VA', 's3supplycounter': 4904.5087, 's3supplycounterunit': 'kVAh', 'q3consume': 308.9, 'q3consumeunit': 'VAr', 'q3consumecounter': 12016.833, 'q3consumecounterunit': 'kVArh', 'q3supply': 0.0, 'q3supplyunit': 'VAr', 'q3supplycounter': 24857.1294, 'q3supplycounterunit': 'kVArh', 'i3': 1.498, 'i3unit': 'A', 'u3': 228.732, 'u3unit': 'V', 'cosphi3': 0.989, 'cosphi3unit': '°', 'speedwire-version': '1.2.4.R|010204'}
2022-02-01 09:51:33: PID: 4743: root: Beginning update
2022-02-01 09:51:33: PID: 4743: root: SMA serials: 1900210416
2022-02-01 09:51:33: PID: 4743: legacy run server: Received command ["modules.sma.device","counter","1900210416"]`

LKuemmel, 2022-02-01:
Ich habe das in PR #1981 gefixt. Bitte nochmal probieren.

MartinRinas, 2022-02-01:
sehr gut, danke! Beide Szenarien laufen nun wieder, sowohl mit Seriennummer 'none' als auch mit Seriennummer. danke!

closed 2022-02-01 by MartinRinas