snaptec/openWB #2008: Update schlägt manchmal fehl

yankee42, 2022-02-08:
Manchmal, ohne dass ich irgend ein Muster erkennen kann, schlägt das Update fehl. Im debug-Log sehe ich dann sowas:
```
atreboot.sh started
loading config
2022-02-08 10:33:01: Update in progress (LV0) at 36 main /var/www/html/openWB/regel.sh
isk /var/www/html/openWB/ramdisk
kill: (32170): No such process
```
Danach kommen nurnoch Ausgaben die sagen, dass gerade ein Update in Progress ist und es passiert nichts mehr. Wenn das passiert führe ich folgendes aus:
```bash
echo 0 > /var/www/html/openWB/ramdisk/bootinprogress; echo 0 > /var/www/html/openWB/ramdisk/updateinprogress
```
Dann bekomme ich die UI wieder und führe das Update nochmal aus. Auf den zweiten Versuch hat es bisher immer geklappt.

Möglicherweise gab es genau deswegen irgendwann schonmal #951, leider fehlt dem Ticket jeglicher Text um zu erfahren, was man damals wusste. Der Fix von damals hilft jedenfalls nicht. Demo:

Erstelle eine Datei `tmp.sh` mit folgendem Inhalt:
```bash
(sleep 1; touch /tmp/demo) &
pkill -f tmp.sh
```
Und führe aus:
```
bash tmp.sh; sleep 2; ls /tmp/demo
Terminated
ls: cannot access '/tmp/demo': No such file or directory
```
Das kill beendet auch die gestartete subshell. Da durch #951 hinzugekommen ist:
```
sudo kill $(ps aux |grep '[a]treboot.sh' | awk '{print $2}'); echo 0 > /var/www/html/openWB/ramdisk/bootinprogress; echo 0 > /var/www/html/openWB/ramdisk/updateinprogress
```
killt sich das Script selbst, bevor es zum `echo` kommt. Was immer da versucht wurde: Es funktioniert nicht.

Eine Idee, die mir gerade kommt: Was ist eigentlich wenn man zweimal auf Update klickt und zwischen den beiden Updates etwa 5 Minuten sind. Dann killt doch der erste Update-Prozess den zweiten?!

benderl, 2022-02-14:
Das ganze `kill $(ps ... grep ...)` könnte durch `kill "$$"` ersetzt werden. Muss man nur darauf vertrauen, dass die PID inzwischen nicht schon von einem anderen Prozess belegt ist.

Sauberer wäre sicher eine Art Watchdog im atreboot.sh.

benderl, 2022-02-23:
#2054

closed 2022-02-23 by benderl