snaptec/openWB #2022: PV Charge Mode incl. Hybrid Systems

ChrisMuki, 2022-02-10:
Consider Hybrid Systems for limiting possible Battery discharge

discharge battery only if SOC>MinSOC and necessary to keep charge going

ChrisMuki, 2022-02-10:
Hallo Leute, ich würde den PV ladungsmodus gerne anpassen und Hybridsysteme berücksichtigen.
Weiter sind noch 1-2 kleine Bugs vorhanden, die in ganz speziellen Situationen auftreten.

Die Änderungen sind überschaubar. Ich möchte nur wissen wann und auf welchen commit basierend ich die Änderung als Pull Request stellen soll. Ist der aktuelle Master korrekt, da ja nun ne neue Beta draußen ist und die fixed dann dort passieren?

benderl, 2022-02-11:
Ja, immer den aktuellen master Branch als Basis nehmen.

closed 2022-02-24 by ChrisMuki