snaptec/openWB #2024: Fehlerhafte Berechnung der Kosten im Ladelog?

cgiesche, 2022-02-11:
Kernel: Linux 4.19.66-v7+ GNU/Linux
openWB Version: 1.9.245

Hallo zusammen,

ich habe in den Einstellungen meinen Strompreis konfiguriert, jedoch wir im Ladelog offensichtlich trotzdem mit einem anderen kWh-Preis (Faktor 0,2999 statt dem konfigurierten 0,3268) gerechnet, so dass die angezeigten Beträge deutlich unter den tatsächlichen liegen.

Übersehe ich irgendetwas bei der (eigentlich nicht sonderlich komplizierten) Konfiguration?

[image]

[image]

AndreasRauhut, 2022-02-11:
Es bringt Dich wahrscheinlich nicht weiter, aber mit der aktuellen Nightly Version: 1.9.255 (2022-02-10 11:58:39 +0100 [b366235e] ) stimmen bei mir die Werte exakt.

cgiesche, 2022-02-11:
Ich werd's einfach mal riskieren. Es gibt ja notfalls einen Weg zurück.

cgiesche, 2022-03-23:
Das Problem hat sich (spätestens) ab Version 1.9.257 erledigt.