snaptec/openWB #2028: u1p3pcheck.sh Bug

hhoefling, 2022-02-11:
Beim ansehen kommt mit folgendes komisch vor,

Bei LP 4-8 wird der Stromwunsch an 4 gesendet.

```
	if [[ $lastmanagementlp4 == 1 && $evseconlp4 == "extopenwb" ]]; then
		oldlllp4=$(<ramdisk/llsolllp4)
		echo $oldlllp4 > ramdisk/tmpllsolllp4
		mosquitto_pub -r -t openWB/set/isss/Current -h $chargep4ip -m "0"
	fi
	if [[ $lastmanagementlp5 == 1 && $evseconlp5 == "extopenwb" ]]; then
		oldlllp5=$(<ramdisk/llsolllp5)
		echo $oldlllp5 > ramdisk/tmpllsolllp5
		mosquitto_pub -r -t openWB/set/isss/Current -h $chargep4ip -m "0"
	fi
	if [[ $lastmanagementlp6 == 1 && $evseconlp6 == "extopenwb" ]]; then
		oldlllp6=$(<ramdisk/llsolllp6)
		echo $oldlllp6 > ramdisk/tmpllsolllp6
		mosquitto_pub -r -t openWB/set/isss/Current -h $chargep4ip -m "0"
	fi
	if [[ $lastmanagementlp7 == 1 && $evseconlp7 == "extopenwb" ]]; then
		oldlllp7=$(<ramdisk/llsolllp7)
		echo $oldlllp7 > ramdisk/tmpllsolllp7
		mosquitto_pub -r -t openWB/set/isss/Current -h $chargep4ip -m "0"
	fi
	if [[ $lastmanagementlp8 == 1 && $evseconlp8 == "extopenwb" ]]; then
		oldlllp8=$(<ramdisk/llsolllp8)
		echo $oldlllp8 > ramdisk/tmpllsolllp8
		mosquitto_pub -r -t openWB/set/isss/Current -h $chargep4ip -m "0"
	fi
```

Frage können LP4-8 weg ?
Wurden ja anscheinden bisher noch nie genutzt

hhoefling, 2022-02-11:
Vergssen....
Beim Subcommande "Start" und "Stop"

benderl, 2022-02-14:
Sollte damit behoben werden: https://github.com/snaptec/openWB/pull/2031

closed 2022-02-14 by benderl