snaptec/openWB #2041: Nach Upgrade von 245 auf 254 ist Web GUI fehlerhaft

hannesrh, 2022-02-17:
Kernel: Linux 5.10.63-v7+ GNU/Linux
Python 3.7.3
openWB Version: 1.9.254

Alle Daten werden über den ioBroker / Node Red / MQTT an die openWB standalone übertragen.
Seit dem Upgrade von 245 auf 254 ist die Anzeige fehlerhaft.

1. Bei der PV Leistung fehlt die Einheit W, außer die PV Leistung ist 0, dann steht dort 0 W
2. Die PV Leistung wird wie in der MQTT Schnittstellen Beschreibung als openWB/set/pv/1/W PV-Leistung in Watt, int, negativ übertragen. Hat sich da etwas geändert?
3. Die geladene Leistung in der GUI oben und beim Ladepunkt sind unterschiedlich, warum?
4. 
[image]

yankee42, 2022-02-17:
Ja, da hat sich was geändert durch #1879 + #1942. Vorher wurde das Vorzeichen komplett ignoriert, jetzt wird ein positives Vorzeichen erwartet. Eigentlich war es garnicht meine Absicht das Verhalten dort zu ändern. Ich glaube ich hatte garnicht realisiert, dass das Vorzeichen komplett ignoriert wurde. In #2030 wurde auch die Dokumentation dazu angepasst.

Ich finde es eigentlich sinnvoller/logischer das Vorzeichen nicht zu ignorieren, da ansonsten eine übertragene Standby-Last als Erzeugung gewertet wird. Allerdings ist die Änderung so natürlich ein "breaking change". Muss @LKuemmel oder @benderl was zu sagen ob das "won't fix" ist oder da wieder absolute Werte genommen werden sollen.

hhoefling, 2022-02-17:
Da bin ich auch schon drauf reingefallem-
Kompatiblität ade.

vorher stand da mal 
```
                if (float(msg.payload) >= -10000000 and float(msg.payload) <= 100000000):
                    if (float(msg.payload) > 1):
                        pvwatt=int(float(msg.payload.decode("utf-8"))) * -1
                    else:
                        pvwatt=int(float(msg.payload.decode("utf-8")))
```
as einem abs() gleich kommt.
Also wurde der negative-Absolutbetrag in die Ramdisk geschreiben egal was auf das mqtt topic geschrieben wurde.

hhoefling, 2022-02-17:
Wenn ich dran denke mit welcher velementz mein 80=80% Änderungsvorschlag abgeschmettert wurde dann erstaunt mich die einfachheit dieser verschlimmbesserung doch schon ein wenig.

benderl, 2022-07-07:
Die Anzeige sollte mit der aktuellen Version wieder passen. Wenn es noch zu Problemen kommt, bitte ein neues Issue öffnen.

closed 2022-07-07 by benderl