snaptec/openWB #2049: 24488883

dj3mu, 2022-02-22:


benderl, 2022-02-23:
Yes, after porting the code to python, the startup of each call takes some seconds. We already refactored several modules to be called by the new legacy_run_server.py. This server caches all imports and eliminates the long startup timeout.

This is still work in progress and refactoring all modules will take some more time. Until then, I would suggest to slow down the regulation to "langsam" which will skip every second loop resulting in 20 seconds for each call.

dj3mu, 2022-02-23:
OK. Thanks for reply and sharing the plannings.  
I've already switched to "langsam" which is fine so far.  
If I can help in any way, don't hesitate to contact me.

benderl, 2022-02-24:
Performance should be improved now. Please try current master.

dj3mu, 2022-02-25:
Perfect. All FEMS queries now done in < 1 s and:

`2022-02-25 06:56:06: PID: 20568: **** Regulation loop needs 4 seconds (LV2) at 55 cleanup /var/www/html/openWB/regel.sh`

I can run again with "Normal" control loop interval.

Thanks for quick integration of the required changes.

closed 2022-02-25 by dj3mu

MiOrt, 2022-02-25:
Not sure if that's the same issue... just updated to last nightly:

`2022-02-25 08:28:11: PID: 31586: **** FATAL ********************************* (LV0)
2022-02-25 08:28:11: PID: 31586: **** FATAL Regulation loop needs 10 seconds (LV0)
2022-02-25 08:28:11: PID: 31586: **** FATAL ********************************* (LV0)
2022-02-25 08:25:04: PID: 7256: ##### cron5min.sh finished ##### (LV0)
2022-02-25 08:25:02: PID: 7256: ##### cron5min.sh started ##### (LV0)
2022-02-25 08:23:01: PID: 22147: **** FATAL ********************************* (LV0)
2022-02-25 08:23:01: PID: 22147: **** FATAL Regulation loop needs 10 seconds (LV0)
2022-02-25 08:23:01: PID: 22147: **** FATAL ********************************* (LV0)
2022-02-25 08:20:03: PID: 31492: ##### cron5min.sh finished ##### (LV0)
2022-02-25 08:20:02: PID: 31492: ##### cron5min.sh started ##### (LV0)
2022-02-25 08:17:50: PID: 12745: **** FATAL ********************************* (LV0)
2022-02-25 08:17:50: PID: 12745: **** FATAL Regulation loop needs 9 seconds (LV0)
2022-02-25 08:17:50: PID: 12745: **** FATAL ********************************* (LV0)`

benderl, 2022-02-25:
Seems like the warning is not thrown every loop but something around every 5 Minutes. Maybe related to something else triggered in this intervall? Awattar / Tibber update?

Please use debug level 2 for further investigation and open a separate issue for this.

MiOrt, 2022-02-25:
No Awattar/Tibber active anymore, will investigate and open new issue

dj3mu, 2022-02-25:
I'm also seeing the "FATAL" intermittently. It happens when other scripts (mainly cron5min) are running.
For me this is acceptable.

But I'm seeing another issue with latest nightly: PV and Min+PV charging is not working. Charging never starts!  
All values of PV, EVU-Kit, Battery are shown properly in chart. So it doesn't seem to be a problem of data retrieval.

Maybe I open a new issue for this (including debug traces) when I find time. Or is it a known issue that is already under investigation?

benderl, 2022-02-25:
We are not aware of any general issue.

MiOrt, 2022-02-25:
Here also no problem, PV and Min+PV charging is working as expected on latest nightly

MiOrt, 2022-03-01:
I have to correct my last posting... there actually seems to be a problem at least with non-starting PV-charging. Will open an issue on that.