snaptec/openWB #2160: Feature request TWCManager-Modul:  Anzeige geladene kWh/km

BillGOH, 2022-04-24:
Mittlerweile funktioniert die Anzeige der geladenen kWh im grauen Feld auf der openWB-Startseite. openWB kann jetzt mit der HTTP-Ansteuerung die Gesamtzahl der kWh des Tesla Wallconnectors über den TWCManager auslesen und damit die geladenen kWh berechnen.

Allerdings werden die geladenen kWh in der Statuszeile des Ladepunkts nicht angezeigt/berechnet. Dort steht immer: "geladen 0,0 kWh / 0,0 km". Das müsste sich doch recht einfach anpassen lassen?

Danke!

BillGOH, 2022-04-27:
Die Berücksichtigung der vom TWCManager rückgemeldeten Gesamt-kWh scheint nicht richtig zu funktionieren. Zum einen sieht der Ladelog öfter so aus:

[image]

Zum anderen sieht die Übersichtsanzeige heute so aus (mit exorbitantem Hausverbrauch):

[image]

Mein System: openWB 1.9.264 auf Raspberry Pi 3B+ (Kernel: Linux 5.10.103-v7+ GNU/Linux, Python 3.7.3) mit TWC-Manager 1.3.0, Fronius Symo GEN24 10 plus, BYD HVM 22.1

benderl, 2022-04-29:
Dann sollte sich das jemand ansehen, der den TWC Manager nutzt. Die openWB denkt sich ja keine zufälligen Zählerstände aus. ;-)

BillGOH, 2022-04-30:
Konnte das Problem nach einigem Probieren selbst lösen:

Ich schalte den Tesla Wallconnector über Homelink automatisch ab, wenn ich den Carport verlasse. Dann liefert der TWCManager als Gesamtladeleistung "0", was openWB verwirrt. Lösung: In der main.sh unter /var/www/html/openWB/modules/twcmanagerlp1 Folgendes "umbauen":

`echo $kwh_total > /var/www/html/openWB/ramdisk/llkwh`
ersetzen durch
```
if [[ $kwh_total != 0 ]]; then
	echo $kwh_total > /var/www/html/openWB/ramdisk/llkwh
fi
```

Damit openWB während des Ladevorgangs mitrechnet und auch der Status als angesteckt erkannt wird (SoC-Modul!) Folgendes in der main.sh ergänzen:

`car=$(echo "$slave" | jq 'first(.[].carsCharging)')`

```
#Ladestatus
if [[ $car == 1 ]] ; then
	echo 1 > /var/www/html/openWB/ramdisk/chargestat
	echo 1 > /var/www/html/openWB/ramdisk/plugstat
else
	echo 0 > /var/www/html/openWB/ramdisk/chargestat
	echo 0 > /var/www/html/openWB/ramdisk/plugstat
fi
```

Vielleicht kann das einer der Profis einpflegen? Hinweis: Geht nur wenn "HTTPControl / Ngardiner Fork" aktiviert ist.

soso1999, 2022-08-09:
Hi BillGOH
Welchen main.sh meinst du genau? Kannst du bitte den Pfad der Datei angeben? 
Danke!

BillGOH, 2022-08-09:
Hallo,
der Pfad ist /var/www/html/openWB/modules/twcmanagerlp1.

soso1999, 2022-08-10:
Perfekt! Vielen Dank!

benderl, 2022-08-11:
Wie wäre es, wenn ihr einen PR mit den Anpassungen stellt?