snaptec/openWB #2178: Wrong calculation of Hausverbrauch when 2nd PV source is available

alfredlupper, 2022-05-04:
[image]
[image]
[image]

Values from Fronius Symo Inverter, 1st PV source with S0 Meter.

{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 122,
               "E_Day" : 9520,
               "E_Total" : 12143.1513671875,
               "E_Year" : 1846039,
               "P" : 4151
            }
         },
         "Site" : {
            "E_Day" : 9520,
            "E_Total" : 12143.15111111111,
            "E_Year" : 1846039,
            "Meter_Location" : "load",
            "Mode" : "vague-meter",
            "P_Akku" : null,
            "P_Grid" : -3925.9450622434929,
            "P_Load" : -225.05493775650712,
            "P_PV" : 4151,
            "rel_Autonomy" : 100,
            "rel_SelfConsumption" : 5.4217041136233952
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2022-05-04T12:42:27+02:00"
   }
}

Power from 2nd Inverter:

{"power":1116.5}

-----------------------------------------------------------------------------------------------------------

Hello,

I found the following issue with Hausverbrauch shown on Dashboard:

The Hausverbrauch value on my OpenWB dashboard is calculated wrongly.
The Hausverbrauch shown is actually wrongly calculated from the correct load value read out from Fronius Symo P_Load
(see Json above) plus the power value of my second inverter! {"power":1116.5}
See picture of the dashboard and my configuration.

Values do not exactly match due to the time difference of the captures.
I have tried it out, if I remove the 2nd PV source the Hausverbrauch is correctly shown.
The power of the 2nd PV source is just added to the real measured Hausverbrauch.

Please check and fix it.

Br,

Fred

vuffiraa72, 2022-05-15:
Hi,

that is an interesting constellation, but I am not sure if it is a simple software problem.

What are the facts. Your S0-counter is installed in consumption path (Hausverbrauch). Further the Fronius inverter measure the solar production and it calculates feed-in from known value (consumption, production):
- "P_Grid" : -3925.9450622434929, (calculated)
- "P_Load" : -225.05493775650712, (measured)
- "P_PV" : 4151, (measured)

OpenWB requires an EVU modul to do its charge control. It needs the data measured at feed-in point and the actual solar production. With this data your consumption is calculated.

Here is the issue. OpenWB relies on calculated feed-in data, which is not taking your second inverter into account. But on the other hand OpenWB knows the solar production from both of your inverters. With that logic all remaining energy has to be consumed.

From a hardware perspective there are two options:
- install a supported counter at feed-in point (OpenWB EVU Kit)
- integrate the second inverter into the Fronius universe (not sure if Fronius supports this)

Regards,
Ulf

tobstare, 2022-06-03:
Hi,

I have a similar problem. I don't know if this has something to do with this, but my home consumption is incorrect because the sum of the two inverters is incorrect.
[image]

Kostal Plenticore plus 7.0 with BYD HV 5.0 + Kostal Piko 15 and an KOSTAL Smart Energy Meter.
The image above is from the KOSTAL Smart Energy Meter.

Regards,
Tobias

tobstare, 2022-06-03:
Might be relevant.
The upper value is used by the KOSTAL Smart Energy Meter, the lower possibly by openWB?
[image]

tobstare, 2022-06-03:
Ich habe gerade mal in den Code geschaut. 
'dxsEntries', ['33556736', '251658753', '83887106', '83887362', '83887618']
Hier scheint mir der erste Wert falsch zu sein. Hier sollte '67109120' stehen. Das wäre dann der "GridAC" und somit zumindest meiner Meinung nach der richtige Wert. Es hilft ja nichts zu wissen, was die Panales liefern. Der Output des Wechselrichters ist für uns interessant, oder?

  -Total DC input [W][double] --> http://192.168.178.21/api/dxs.json?dxsEntries=33556736
  -Output power [W][double] --> http://192.168.178.21/api/dxs.json?dxsEntries=67109120

Somit hat mein "Fehler" aber wohl nichts mit dem obigen zu tun, sorry.