snaptec/openWB #2211: Plenticore keine Daten

cvastag, 2022-05-18:
Hallo zusammen,

leider bekomme ich keine Datenübermittlung vom Plenticore zur OpenWB hin. 
Meine OpenWB läuft über Proxmox auf einem Intel Nuc.
Die Komponenten KSEM und Plenticore und die OpenWB sind alle über Kabel mit einem Ubiquiti Switch verbunden. Davor hängt eine UDR Station als Router

Im Plenticore und KSEM ist TCP und Modbus aktiviert
Als PV Modul ist der Plenticore mit der IP Eingetragen
Als EVU Plenticore mit KSEM
Trage ich das KSEM als EVU ein mit der IP, erhalte ich auch keine Daten.

Jemand vielleicht noch ein Tipp? Kann es vielleicht an iwelchen Firewall regeln liegen?

skl77, 2022-05-18:
Kannst du via proxmox die Webuu von plenticore oder KSM erreichen ?

benderl, 2022-05-18:
Solche Themen sind im Forum besser aufgehoben.
Ich bin sicher, dass Dir dort schneller geholfen wird.

closed 2022-05-18 by benderl