snaptec/openWB #2215: Syntax Fehler bei mypeugeot SOC Ermittlung

Digitalvitamin, 2022-05-21:
Verbaut ist eine series2 Standard+, SW-Version: 1.9.262 
Verwendet wird das MyPeugeot SOC Modul zum Abruf des SOC eines e208.

Im Debug-Log sieht man beim SOC-Abruf verschiedene Fehlermeldungen:

```
2022-05-21 10:54:13: PID: 8826: ungültiger Wert für soc:  (LV0)
2022-05-21 10:53:53: PID: 7415: ungültiger Wert für soc:  (LV0)
(standard_in) 2: syntax error
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.1' kann nicht angelegt werden: Keine Berechtigung
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.2' kann nicht angelegt werden: Keine Berechtigung
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.3' kann nicht angelegt werden: Keine Berechtigung
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.4' kann nicht angelegt werden: Keine Berechtigung
TypeError: list indices must be integers, not str
    fetchedsoctime = batt['energy'][0]['updatedAt']
  File "/var/www/html/openWB/modules/soc_mypeugeot/peugeotsoc.py", line 128, in <module>
Traceback (most recent call last):
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.1' kann nicht angelegt werden: Keine Berechtigung
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.2' kann nicht angelegt werden: Keine Berechtigung
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.3' kann nicht angelegt werden: Keine Berechtigung
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.4' kann nicht angelegt werden: Keine Berechtigung
2022-05-21 10:50:04: PID: 20672: ##### cron5min.sh finished ##### (LV0)
2022-05-21 10:50:02: PID: 20672: ##### cron5min.sh started ##### (LV0)
modules/soc_mypeugeot/main.sh: Zeile 120: 1653047233 - : Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \"- \").
modules/soc_mypeugeot/main.sh: Zeile 120: /var/www/html/openWB/ramdisk/peugeotsoctime: Datei oder Verzeichnis nicht gefunden
TypeError: list indices must be integers, not str
    fetchedsoctime = batt['energy'][0]['updatedAt']
  File "/var/www/html/openWB/modules/soc_mypeugeot/peugeotsoc.py", line 128, in <module>
Traceback (most recent call last):
cp: reguläre Datei '/var/www/html/openWB/openwb.conf.1' kann nicht angelegt werden: Keine Berechtigung
```

Im EV SOC-Log sieht das wie folgt aus:
```
2022-05-21 10:52:37: PID: 2003: Lp1: Fetched from myPeugeot: 66% (LV0)
2022-05-21 10:52:33: PID: 2003: Lp1: Ladestatus changed to charging. Fetching SoC from myPeugeot out of order. (LV0)
2022-05-21 10:47:53: PID: 4416: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-21 10:37:43: PID: 23766: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-21 10:27:33: PID: 10873: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-21 10:17:24: PID: 30378: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-20 14:38:04: PID: 4401: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-20 14:27:53: PID: 23874: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-20 14:17:43: PID: 11029: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-20 14:07:33: PID: 30452: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-20 13:57:23: PID: 17615: Lp1: Fetching SoC from myPeugeot (LV0)
2022-05-20 10:11:37: PID: 24135: Lp1: Fetched from myPeugeot: 53% (LV0)
2022-05-20 10:11:33: PID: 24135: Lp1: Ladestatus changed to charging. Fetching SoC from myPeugeot out of order. (LV0)
2022-05-20 10:08:33: PID: 875: Lp1: Fetching SoC from myPeugeot (LV0)
```

SOC kann nur kurz nach dem Start eines Ladevorgangs ausgelesen werden. Das führt dazu dass beim Anstecken des Fahrzeugs die openWB immer noch "denkt" der SOC wäre bei 80% und startet nicht mit dem Laden.

snaptec, 2022-05-21:
Tritt das Problem denn auch in der aktuellen Version auf?

Digitalvitamin, 2022-05-21:
Sowohl laut Github History als auch laut Changelog im Forum hat sich doch an dem Modul nichts geändert, oder?
(Aufgrund der [diversen Probleme](https://www.openwb.de/forum/viewtopic.php?f=9&t=5237) mit der 1.9.265 habe ich mich noch nicht getraut ein Update zu machen.)

benderl, 2022-05-24:
Bitte im Support-Beitrag nachfragen, da die SoC Module von der Community gepflegt werden.
https://www.openwb.de/forum/viewtopic.php?f=12&t=3139

closed 2022-05-24 by benderl