snaptec/openWB #2216: Debug Log - exception when Inverter goes into sleep mode

ChristophCaina, 2022-05-21:
Hi,
I've noticed some error messages when my inverter is in sleep mode (during night)

```
2022-05-21 23:15:02: PID: 30903: ##### cron5min.sh started ##### (LV0)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.***.***.***:502)]
raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
File "/usr/local/lib/python3.5/dist-packages/pymodbus/client/sync.py", line 119, in __enter__
self.delegate.__enter__()
File "/var/www/html/openWB/packages/modules/common/modbus.py", line 54, in __enter__
with ModbusClient(ipaddress, port=modbusport) as client:
File "/var/www/html/openWB/modules/bezug_solaredge/solaredge.py", line 27, in update
parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
args.RUN(args)
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 34, in run_using_positional_cli_args
run_using_positional_cli_args(update, argv)
File "/var/www/html/openWB/modules/bezug_solaredge/solaredge.py", line 69, in main
importlib.import_module(parsed[0]).main(parsed[1:])
File "/var/www/html/openWB/packages/legacy_run_server.py", line 109, in handle_message
self.__callback(read_all_bytes(connection))
File "/var/www/html/openWB/packages/legacy_run_server.py", line 87, in handle_connection
yield
File "/var/www/html/openWB/packages/legacy_run_server.py", line 48, in redirect_stdout_stderr_exceptions_to_log
Traceback (most recent call last):
2022-05-21 23:13:43: PID: 22992: legacy run server: Unhandled exception
2022-05-21 23:13:43: PID: 22992: pymodbus.client.sync: Connection to (192.***.***.***, 502) failed: [Errno 111] Connection refused
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.***.***.***:502)]
raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
File "/usr/local/lib/python3.5/dist-packages/pymodbus/client/sync.py", line 119, in __enter__
self.delegate.__enter__()
File "/var/www/html/openWB/packages/modules/common/modbus.py", line 54, in __enter__
with ModbusClient(ipaddress) as client:
File "/var/www/html/openWB/modules/wr_solaredge/solaredgeall.py", line 98, in update
parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
args.RUN(args)
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 34, in run_using_positional_cli_args
run_using_positional_cli_args(update, argv)
File "/var/www/html/openWB/modules/wr_solaredge/solaredgeall.py", line 111, in main
importlib.import_module(parsed[0]).main(parsed[1:])
File "/var/www/html/openWB/packages/legacy_run_server.py", line 109, in handle_message
self.__callback(read_all_bytes(connection))
File "/var/www/html/openWB/packages/legacy_run_server.py", line 87, in handle_connection
yield
File "/var/www/html/openWB/packages/legacy_run_server.py", line 48, in redirect_stdout_stderr_exceptions_to_log
Traceback (most recent call last):
2022-05-21 23:13:42: PID: 22992: legacy run server: Unhandled exception
2022-05-21 23:13:42: PID: 22992: pymodbus.client.sync: Connection to (192.***.***.***, 502) failed: [Errno 111] Connection refused
2022-05-21 23:13:42: PID: 19264: **** FATAL ********************************* (LV0)
2022-05-21 23:13:42: PID: 19264: **** FATAL Regulation loop needs 11 seconds (LV0)
2022-05-21 23:13:42: PID: 19264: **** FATAL ********************************* (LV0)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.***.***.***:502)]
raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
File "/usr/local/lib/python3.5/dist-packages/pymodbus/client/sync.py", line 119, in __enter__
self.delegate.__enter__()
File "/var/www/html/openWB/packages/modules/common/modbus.py", line 54, in __enter__
with ModbusClient(ipaddress, port=modbusport) as client:
File "/var/www/html/openWB/modules/bezug_solaredge/solaredge.py", line 27, in update
parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
args.RUN(args)
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 34, in run_using_positional_cli_args
run_using_positional_cli_args(update, argv)
File "/var/www/html/openWB/modules/bezug_solaredge/solaredge.py", line 69, in main
importlib.import_module(parsed[0]).main(parsed[1:])
File "/var/www/html/openWB/packages/legacy_run_server.py", line 109, in handle_message
self.__callback(read_all_bytes(connection))
File "/var/www/html/openWB/packages/legacy_run_server.py", line 87, in handle_connection
yield
File "/var/www/html/openWB/packages/legacy_run_server.py", line 48, in redirect_stdout_stderr_exceptions_to_log
Traceback (most recent call last):
2022-05-21 23:13:39: PID: 22992: legacy run server: Unhandled exception
2022-05-21 23:13:39: PID: 22992: pymodbus.client.sync: Connection to (192.***.***.***, 502) failed: timed out
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.***.***.***:502)]
raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
File "/usr/local/lib/python3.5/dist-packages/pymodbus/client/sync.py", line 119, in __enter__
self.delegate.__enter__()
File "/var/www/html/openWB/packages/modules/common/modbus.py", line 54, in __enter__
with ModbusClient(ipaddress) as client:
File "/var/www/html/openWB/modules/wr_solaredge/solaredgeall.py", line 98, in update
parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
args.RUN(args)
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 34, in run_using_positional_cli_args
run_using_positional_cli_args(update, argv)
File "/var/www/html/openWB/modules/wr_solaredge/solaredgeall.py", line 111, in main
importlib.import_module(parsed[0]).main(parsed[1:])
File "/var/www/html/openWB/packages/legacy_run_server.py", line 109, in handle_message
self.__callback(read_all_bytes(connection))
File "/var/www/html/openWB/packages/legacy_run_server.py", line 87, in handle_connection
yield
File "/var/www/html/openWB/packages/legacy_run_server.py", line 48, in redirect_stdout_stderr_exceptions_to_log
Traceback (most recent call last):
2022-05-21 23:13:35: PID: 22992: legacy run server: Unhandled exception
2022-05-21 23:13:35: PID: 22992: pymodbus.client.sync: Connection to (192.***.***.***, 502) failed: timed out
```

since these errors starting in the evening, I think, that this is when the inverter is shutting down;

My Inverter is SolarEdge - and from my previous Monitoring through Modbus, I know that the Inverter is sending its current status

[image]

I think, it could be an idea to get the inverter status, if the error is related to the fact that the inverter is sleeping ...

LKuemmel, 2022-05-24:
Do you know which register I have to query and how the data is formatted (UINT16, ..)?

ChristophCaina, 2022-05-25:
Hi, I've had a look into the specifications provided by solar edge...
I think, this is what you're looking for?

[image]
[https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf](https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf)

ChristophCaina, 2022-05-27:
Hi @LKuemmel, ein kurzes Update:

Ich habe bei mir das Setup ein wenig verändert.
Die OpenWB bekommt Ihre Daten nun nicht mehr vom Inverter direkt, sondern von einer modbus-bridge, welche mit dem Inverter verbunden ist.
Das hat den Vorteil, dass ich noch aus anderen Anwendungen den Inverter monitoren kann.

Leider treten auch jetzt noch diese Fehler in den Logs auf - zuletzt abends gegen 23 Uhr...
Mir ist nicht ganz klar, was jetzt genau dazu führen sollte, dass die OpenWB immer wieder zu den gleichen Uhrzeiten Verbindungsprobleme mit der Gegenstelle bekommt...
Meine Monitoring-Lösung scheint hier keinerlei probleme zu haben, also dürfte es auch nicht so sein, dass der Inverter dann die Kommunikation einstellt... (?)