snaptec/openWB #2219: Insufficient handling of URL components in modules

DetMoerk, 2022-05-24:
Das Json Modul macht Probleme bei Shelly als Inverter. Es hat die Komponente path gefehlt.

 parsed_url = parse_url(ip_address)```

    device_config = get_default_config()
    device_config["configuration"]["protocol"] = parsed_url.scheme
    device_config["configuration"]["domain"] = parsed_url.hostname
    if parsed_url.port is None:
        device_config["configuration"]["port"] = 80
    else:
        device_config["configuration"]["port"] = int(parsed_url.port)
    device_config["configuration"]["path"] = parsed_url.path
```

```
def __init__(self, device_config: dict) -> None:
        self.components = {}  # type: Dict[str, json_component_classes]
        try:
            self.device_config = device_config
            port = self.device_config["configuration"]["port"]
            path = self.device_config["configuration"]["path"]
            if path is None:
                path= '/'
            self.domain = self.device_config["configuration"]["protocol"] + \
                "://" + self.device_config["configuration"]["domain"] + \
                ":" + str(port) + path
        except Exception:
            log.exception("Fehler im Modul "+device_config["name"])

```
Ich hab das jetzt mal bei mir so angepasst, damit laufen zumindest VZlogger und Shelly wieder. Wobei die Port=80 Ersetzung natürlich auch schief gehen kann, wenn z.B. https:// als Protokoll angegeben ist. Was auch noch fehlt. wären Query Parameter , die mit ? hinten angehängt werden. Das macht der parse_url jetzt auch kaputt, da diese nicht ermittelt und in die Konfiguration übertragen werden. Authentifizierung mit

http://user:password@meineDomain.de/
ist ebenfalls nicht berücksichtigt. Ich hab jetzt nicht explizit geschaut, aber ich vermute bei http Modulen (und anderen auf http basierten Modulen) könnte das jetzt mit parse_url ebenfalls Probleme geben.

```
pi@raspi4:/var/www/html/openWB/packages $ grep -R parse_url
modules/http/device.py:from urllib3.util import parse_url
modules/http/device.py:    parsed_url = parse_url(url)
modules/common/modbus.py:from urllib3.util import parse_url
modules/common/modbus.py:        parsed_url = parse_url(address)
modules/json/device.py.org:from urllib3.util import parse_url
modules/json/device.py.org:    parsed_url = parse_url(ip_address)
modules/json/device.py:from urllib3.util import parse_url
modules/json/device.py:    parsed_url = parse_url(ip_address)
```

DetMoerk, 2022-05-25:
Wäre es nicht möglich, das durch parse_url erzeugte URL Objekt in die configuration zu übergeben und mit diesem auch den Request zu erzeugen. Ich denke der parse_url wurde eingeführt, um bei fehlerhaften Benutzereingaben einen Fehler zu werfen.

VG
Det

benderl, 2022-05-25:
Hatte ich mir auch so überlegt. Am einfachsten ist es wohl, die einzelnen Teile (Protokoll, Host, Port) in einem Eingabefeld zu erfassen und nicht aufzuteilen. Dann könnte es mit parse_url verarbeitet werden.

DetMoerk, 2022-05-25:
Das ist ja schon so. Im `parsed_url = parse_url(ip_address)` kommt ja die komplette URL in der Variablen ip_address an, wird dann zerlegt, aber leider werden nicht alle Teile in die configuration übergeben. Aus den Einzelteilen wird dann über

```
self.domain = self.device_config["configuration"]["protocol"] + \
                "://" + self.device_config["configuration"]["domain"] + \
                ":" + str(port) + path
```
zusammen gesetzt, leider nicht mit allem, was eine URL enthalten kann.

DetMoerk, 2022-05-29:
Funktioniert wieder. Danke Lena

closed 2022-05-29 by DetMoerk