snaptec/openWB #2245: Support für neuen SMA Hybrid Wechselrichter

JanFellner, 2022-06-12:
Habe gewechselt von einem alten TriPower 10kw und einem Sunny Boy Storage auf den neuen Hybrid Wechselrichter.

Heute die Batterie ungeklemmt, der Wechselrichter selbst läuft schon ein paar Tage.

Sobald der Wechselrichter jetzt aus der Batterie speist erscheint es so als ob der Generator die Leistung bereitstellt.

Wie muss man vorgehen um diesen neuen Wechselrichter anzubinden?

[image]

JanFellner, 2022-06-12:
Übergangslösung bis der TriPower SE in der Liste auftaucht.
Konfiguriert man einen Sunny Boy Storage mit der IP des neuen Hybrid Wechselrichters stimmt zumindest mal die Anzeige wieder.

JanFellner, 2022-06-12:
[modbus_register.zip](https://github.com/snaptec/openWB/files/8886541/modbus_register.zip)

Die beiden interessanten Modus Register scheinen

- 31393 - momentante Batterie Ladung in W
- 31395 - momentane Batterie Entladung in W

zu sein. Sind diese bei SMA Wechselrichter vorhanden müssen diese auf das Register 30775 angerechnet werden.

Der Wert von 31395 muss von 30775 abgezogen werden um auf die echte PV Leistung zu kommen.

Wie sich das Laden auf den Leistungwert 30775 auswirkt muss ich morgen schauen.

JanFellner, 2022-06-12:
@benderl ich hab leider nicht die passende Umgebung um openWB und das Python zu debuggen, gibts da ne Anleitung wie ihr hier vorgeht wenn ihr entwickelt? Oder soll ich das "trocken" codieren und nen PullRequest machen den ihr dann fein tuned?

(openwb läuft virtualisiert auf nem raspi desktop in esxi. remote debuggen wäre für mich eigentlich die bequemste Variante, am liebsten mit VSCode, ich weiß aber nicht wie Ihr das Projekt pflegt, debugged etc.) Danke ;)

benderl, 2022-06-12:
Naja, die 1.9er kann man schlicht nicht debuggen. Das sind ja zusammengesetzte Shell, Python und andere Skripte. Kannst ja mit VS Code in der VM coden und testen(Logmeldungen). Anders machen wir das mit 1.9 auch nicht.

JanFellner, 2022-06-12:
ich hab mal geforkt, gefummelt und gepusht, bin nicht so der Python Guru :) Ich fummel mich da so durch.

Das dritte Argument im read_holding_registers ist mir unklar unit=1 / unit=3 

https://github.com/snaptec/openWB/pull/2246/commits/4246f5037c472f696b718e17e5ae65ccc4a6aecc

JanFellner, 2022-06-12:
[image]

Für die Batterie Entladung hätten wir nen Fix, ich schau mir morgen mal an wie das beim Batterie laden aussieht und bei den Energiewerten.

hhoefling, 2022-06-12:
> Naja, die 1.9er kann man schlicht nicht debuggen. 

Naja kann man schon!, Haben wir vor 20 Jahren schon so gemacht. :-)

benderl, 2022-06-13:
> > Naja, die 1.9er kann man schlicht nicht debuggen.
> 
> Naja kann man schon! (wenn man kann), Haben wir vor 20 Jahren schon so gemacht. :-)

Das ist aber nicht das, was man heutzutage unter Debugging versteht. Vor 20 Jahren habe ich das auch noch so gemacht. ;-)

benderl, 2022-06-13:
> ich hab mal geforkt, gefummelt und gepusht, bin nicht so der Python Guru :) Ich fummel mich da so durch.
> 
> Das dritte Argument im read_holding_registers ist mir unklar unit=1 / unit=3
> 
> [4246f50](https://github.com/snaptec/openWB/commit/4246f5037c472f696b718e17e5ae65ccc4a6aecc)

Das ist die Modbus-Id (oder Adresse) des auszulesenden Gerätes.

JanFellner, 2022-06-13:
> Das ist die Modbus-Id (oder Adresse) des auszulesenden Gerätes.

Das irritiert mich jetzt etwas denn die Auswahl des Typs zwischen standard und Core ändert doch nichts an den Adressen der Geräte? (Konkreter: Warum hat hier standard eine andere Adresse als der Core 2 Typ)

https://github.com/snaptec/openWB/blob/4246f5037c472f696b718e17e5ae65ccc4a6aecc/packages/modules/sma_modbus_tcp/inverter_modbus_tcp.py#L49

Die Verwendung des Sunny Boy Storage als Workaround zeigt weitere "Schwächen".
Ich würde jetzt auch hier jetzt auch einen SMA Hybrid hinzufügen oder gibt es Vorlagen wo die Daten des Akkus und Batterieladereglers im Kontext des Wechselrichters ausgelesen und übergeben werden?

[image]
[image]

JanFellner, 2022-06-13:
@benderl einfachte Möglichkeit die geänderten Python Scripte zum leben zu erwecken *ohne* openWb neu zu starten? So muss ich immer mehrere Minuten warten bis ich nen neuen Graphen bekomme und die Iterationen dauern damit unnötig lange...

LKuemmel, 2022-06-13:
Mit diesem Skript wird der Python-Server neu gestartet: `./packages/legacy_run_server.sh`
Ich habe am Freitag auch an einer Lösung des Problems gearbeitet, allerdings habe ich bei dem WR aus dem Supportticket aus den Registern 31393 und 31395 immer nur den maximalen UINT32-Wert erhalten. Welches Speicher-Modell hast Du?

hhoefling, 2022-06-13:
> Das ist aber nicht das, was man heutzutage unter Debugging versteht. 
Dann lass die modernen Jungster mal eine Kundenanlage Remote Debugen/Fehlersuchen,
Da brauchst du solche Fähigkeiten wieder.

JanFellner, 2022-06-14:
Was verwendet ihr denn für ein Tool um sich einfach nackt die Modbus Daten ausgeben zu lassen? Ich hab jetzt verschiedene ausprobiert, das ist ja alles sowas von Grottig...
- Chipkin CAS Modbus Scanner
- Simply Modbus Client Test Software
- Novatek Modbus Client....

*KATASTROPHE*....

benderl, 2022-06-14:
> Was verwendet ihr denn für ein Tool um sich einfach nackt die Modbus Daten ausgeben zu lassen? Ich hab jetzt verschiedene ausprobiert, das ist ja alles sowas von Grottig...
> 
> * Chipkin CAS Modbus Scanner
> * Simply Modbus Client Test Software
> * Novatek Modbus Client
> * Modbus Poll
> 
> _KATASTROPHE_....
> 
> Irgendwie suche ich entweder was für die Command line der ich ein TExtfile geben kann oder was mit nem Tabellen Layout wo ich einfach die Register und den Datentyp angeben kann und das Tool liefert mir die Werte...

Spätestens bei den Datentypen wirst Du Abstriche machen müssen. Je nach Typ müssen ja mehrere Register miteinander verknüpft werden. Dann noch little oder big endian oder wordorder. Mir ist da Nichts bekannt. Modbus ist halt ein sehr rudimentäres Protokoll, bei dem nur die Kommunikation standardisiert ist. Bei den Daten kann jeder machen was er will.

JanFellner, 2022-06-17:
Ich habs gestern fertig gestellt und heute früh nochmals nen Blick auf die Daten geworfen. Schaut sauber aus.
PR damit von draft auf ready for review gestellt.

Was mir leider nicht gelungen ist:
- Einen weiteren Speicher mit Namen "SMA Sunny Tripower Smart Energy" hinzuzufügen damit klar ist dass man diesen auch auswählen muss. Nachdem ich hier eine Blaupause vom Sunny Boy Storage gemacht hab ging gar nix mehr. Da fehlt mir in der Tiefe die Kenntnis von openWB. (sind auch irgendwelche php Seiten welche irgendwas mit alt beinhalten -> unklar :))

closed 2022-10-07 by benderl