snaptec/openWB #2270: Fronius Gen24 Hausverbrauch zu hoch da Wandlungsverluste eingerechnet

godfuture, 2022-06-30:
Hi, ich betreibe seit 2020 den Gen24 Plus Symo 6.0. Das Modul für Fronius Gen24 funktioniert. Jedoch weichen die Daten vom Fronius Webfrontend ab.

Der Hausverbrauch ist bei Fronius niedriger. Ich denke, dass OpenWB einfach Produktion minus Einspeisung rechnet. Das würde mit der Angabe in OpenWB übereinstimmen. Jedoch ist der Hausverbrauch niedriger.

In der Annahme, dass der Hausverbrauch in OpenWB berechnet ist...kann das Fronius Modul geändert werden, sodass auch der Hausverbrauch direkt vom WR gelesen wird?

benderl, 2022-06-30:
Nein, der Hausverbrauch wird in openWB immer berechnet. Das macht auch Sinn, da separat ausgewiesene Verbraucher (SmartHome, Ladepunkte) im Hausverbrauch des Wechselrichters enthalten sind.

Der Hausverbrauch in openWB ist das Ergebnis der Verrechnung aller Bezüge (EVU, PV, Speicherentladung) und Verbräuche (Einspeisung, Ladeleistung, SmartHome, Speicherladung).

godfuture, 2022-06-30:
Ich verstehe es noch nicht. Der Hausverbrauch in OpenWB ist klar falsch. Der Fronius Wechselrichter kennt die Verluste und stellt diese korrekt dar. Entsprechend ist der Hausverbrauch im Fronius Webfrontend niedriger.

Da der Fronius WR sich die Daten von einem digitalen Modbus Zähler direkt hinter dem Zweirichtungszähler am Haus holt, sollten andere Ladepunkte und Smart Home richtig von Fronius berechnet werden können.
Warum sollte OpenWB nicht direkt den Hausverbrauch vom WR lesen?

hhoefling, 2022-06-30:
Aber den Verbauch der openWB willst du nicht rausrechnen....(Kopfkratz)
Also für mich sind die Verluste im WR eindeutig "Hausverbrauch",

benderl, 2022-06-30:
Im Hausverbrauch des WR sind alle Ladepunkte etc. mit drin. Dann muss das dort auch wieder verrechnet werden und Die Anzeigen Fronius/openWB sind wieder unterschiedlich.

Wenn Du mal die Summen aus Deinem Screenshot bildest, wirst Du schnell feststellen, dass es beim Fronius nicht passt:
6,19kW PV - 5,80kW EVU Export = 390W != 257W Hausverbrauch

godfuture, 2022-06-30:
> Aber den Verbauch der openWB willst du nicht rausrechnen....(Kopfkratz)

Die OpenWB ist ja ein Gerät in meinem Haus, wieso sollte ich es rausrechnen wollen?

> 6,19kW PV - 5,80kW EVU Export = 390W != 257W Hausverbrauch

Das ist der Punkt. "HausVERBRAUCH", da steckt ja das Wort "Verbrauch" drin. Nur gibt es aber für die fehlenden 133 W keinen Verbraucher. Es ist einfach Verlust. Anders gesagt, wenn ich in OpenWB einfach den Fronius Hausverbrauch anzeige, sehe die korrekt Leistung, die meine Hausnetz verbraucht, und kann in Verrechnung mit der Produktion und Einspeisung sogar die Verluste berechnen. Die Datenqualität bzw. Transparenz steigt. Und ich weiss dann nicht weniger, sondern mehr über mein Hausnetz Bescheid.

benderl, 2022-07-01:
Das ist reine Ansichtssache. Fakt ist, dass nicht bei allen Systemen ein Hausverbrauch auszulesen ist. Daher wird dieser als Differenz aller vorhandenen Messgrößen von openWB errechnet. Das läßt sich nicht für ein spezielles Modul ändern.

closed 2022-07-01 by benderl

godfuture, 2022-07-01:
Die Frage am Ende ist, wieso Fronius die Verluste rausrechnet. Die haben sich bestimmt auch Gedanken gemacht. Aber gut, jetzt ist das Ticket schneller zu als ich Amen sagen kann.

Schade