snaptec/openWB #2283: old data over MQTT

ChristophCaina, 2022-07-06:
I am using the openWB and pulling different information via MQTT to use them in my HomeAssistant Setup.
This will be done by the openWB integration itself (https://github.com/a529987659852/openwbmqtt) and also with NodeRed.

Now, I have noticed some strange behave with the values that are delivered over MQTT.

on a regular basis, when I reboot one of the systems, the values are getting reset to some older values - which have been updated already.

For example:
The last time I have charged my Car, the total amount of charged kWh was near 160 kWh in total.
since then, the value has been reset to 81 kWh ...
[image]
[image]

The interesting thing is, that it will always reset to this value... I have seen it several times between different charges.
[image]

I think, this might be somehow because the OpenWB is not regularely sending out the same Topics when there's no change - and somehow older values have been stored "somewhere"... similar like a cache?
I've noticed this behave with the Integration I've linked above... and tried to workaround this with a node red configuration which connects to the OpenWB directly.

Unfortunately, after it seems to work as expected, today the behave is completely the same - and the values are the same like in the OpenWB HomeAssistant integration.
[image]

[image]

[image]

dj3mu, 2022-07-06:
The "caching" is most likely in the MQTT-Server itself which persists data (infinite) according to the setting of the "Retain" flag of the MQTT message.

To get latest values, try sending a message with raw content `1` to MQTT topic `openWB/set/RenewMQTT`. It should force the openWB to retransmit all MQTT values.

dj3mu, 2022-07-06:
First of all, please also check if the data is really re-transmitted.

Second, please don't send that message every 10 seconds. It will cause significant additional load to the openWB.  
Ideally, you should only and exactly send it once your nodered is restarted.  
Or in which other cases do you encounter this problem?

ChristophCaina, 2022-07-06:
I am not quite sure... probably only when HomeAssistant / NodeRed is restarting.
I will do further tests, but sending the message seemed to work

ChristophCaina, 2022-10-19:
Hi, I need to bring that topic up again.
In the setup mentioned above - I had a MQTT Broker acting as a bridge.
this MQTT broker has received the data from the MQTT Broker of the OpenWB.

I did some further research and noticed, that - in my case - the old values came from my own Broker (at least in the cases that I have observed recently)...

Since then, I've changed the setup.
I am now connected directly to the MQTT Broker in the Wallbox and have removed the bridge in between.

First tests showed, that everything seemed to be OK now - restarts didn't show any old meter values from the OpenWB.

But today - I noticed, that the same situation happened again yesterday evening.
My Smarthome system received an old Meter-Value...
After round about 5 min, the value switched back to the correct value.

Here's how it looked in the history of my Smarthome Software:

[image]
[image]
[image]
[image]

due to the unexpected occurence, I don't have any logs from the OpenWB available - but I will try to do some further research.
At that time, I am not aware of any reboot of the OpenWB during that time frame...

Other people reported similar issues in the forum:
[image]

benderl, 2022-10-19:
Wird im Forum diskutiert, daher schließe ich hier.
https://www.openwb.de/forum/viewtopic.php?f=9&t=5994

closed 2022-10-19 by benderl