snaptec/openWB #2285: sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“

xerion3800, 2022-07-07:
Hallo zusammen,

bekomme seit Tagen folgende Fehlermeldungen im Log. Kann das noch  jemand anderes bestätigen?

```
2022-07-07 08:15:03: PID: 13198: ##### cron5min.sh finished ##### (LV0)
2022-07-07 08:15:01: PID: 13198: ##### cron5min.sh started ##### (LV0)
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
„sleep --help“ liefert weitere Informationen.
........

```

benderl, 2022-07-07:
Kann ich an meinen Systemen nicht nachvollziehen. In der regel.sh gibt es auch nur zwei Stellen, an denen "sleep" genutzt wird. Kannst Du das weiter eingrenzen? Im Debug Mode 1 oder 2 sollte die Stelle identifiziert werden können.

xerion3800, 2022-07-07:
Ich hatte mir die beiden Stellen auch schon angeschaut und nicht was finden könne. Aus Debug Log werde ich auch nicht schlauer:
```
2022-07-07 11:00:02: PID: 6425: legacy_run_server is already running (LV1) at 314 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6425: mqtt handler is already running (LV1) at 301 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6425: current ip: 192.168.178.21 (LV1) at 296 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6425: updating daily yield stats (LV1) at 166 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6425: electricity provider not set up; skipping price update (LV2) at 162 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6425: Netzschutz konfiguriert; aktuelle Frequenz: 0 (LV1) at 111 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6425: daily csv updated: /var/www/html/openWB/web/logging/data/daily/20220707.csv (LV1) at 103 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6425: ##### cron5min.sh started ##### (LV0) at 24 main /var/www/html/openWB/runs/cron5min.sh
2022-07-07 11:00:02: PID: 6426: **** Regulation loop start **** (LV1) at 70 main /var/www/html/openWB/regel.sh
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
„sleep --help“ liefert weitere Informationen.

sleep: sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
„sleep --help“ liefert weitere Informationen.
sleep: ungültiges Zeitintervall „/var/www/html/openWB/regel.sh“
```

benderl, 2022-07-27:
Hast Du neue Infos? Kommen die Meldungen weiterhin auch mit der aktuellen Nightly?

xerion3800, 2022-07-27:
Leider besteht das Problem auch noch mit der aktuellen Nightly, habe gerade frisch aktualisiert. Mehr als aus der Debugausgabe kann ich  leider auch nicht feststellen.

benderl, 2022-07-27:
Ist bei Dir in der openwb.conf igendwie "slavemode" aktiviert worden?
https://github.com/snaptec/openWB/blob/2750c5f8fe055802e8ba4c70aedcbeb15aa4340c/regel.sh#L74

xerion3800, 2022-07-27:
Laut openwb.conf ist slavemode aus:
`slavemode=0`

Ich hatte auch schon mal die sleep variablen durch fixe Zeiten ersetzt hat wohl nichts mit der regel.sh zu tun. Denn wenn sleep einen Fehler meldet, da wird auch der Wert mit ausgebeben. Und in meinem Fall ist der Wert wohl "/var/www/html/openWB/regel.sh" Ich muss jetzt nur herausfinden wo sleep aufgerufen wird, ich vermute es wird ein Zeilenumbruch Fehler sein wodurch der Pfad als Wert angenommen wird.

xerion3800, 2022-07-27:
Habe den Fehler gefunden. Mein Crontab war zerschossen. Sorry

closed 2022-07-27 by xerion3800