snaptec/openWB #2325: Mqtt: verbaute 1p3p Umschaltung ausweisen

andig, 2022-07-25:
Für Nutzung mit externem HEMS wäre es schön, wenn wir erkennen könnten ob die Phasenumschaltung verbaut ist. Aktuell muss ein Anwender dies separat konfigurieren um Phasenumschaltung z.B. mit evcc nutzen zu können.

benderl, 2022-07-27:
Ist leider so, dass die Hardware das nicht hergibt.

andig, 2022-07-27:
Wir löst openWB das dann- Anwendereinstellung? Falls ja: lässt sich die Einstellung über mqtt publizieren?

benderl, 2022-07-27:
Es gibt in 1.9 keine saubere Lösung dafür. In 2.0 ist es besser umgesetzt.

closed 2023-03-13 by benderl

andig, 2023-03-13:
@benderl gibts schon eine Doku, wie das in 2.0 gelöst wird die ich verwenden könnte?

benderl, 2023-03-13:
Nein, da in der 2.0 noch gar keine lokalen Ladepunkte unterstützt werden.