snaptec/openWB #2327: Logging: UnicodeEncodeError: 'ascii' codec can't encode character[..]

yankee42, 2022-07-26:
Ich sehe im Log folgenden Fehler:
```
2022-07-26 09:05:48: at_reboot: --- Logging error ---
2022-07-26 09:05:48: at_reboot: Traceback (most recent call last):
2022-07-26 09:05:48: at_reboot: File "/usr/lib/python3.5/logging/__init__.py", line 983, in emit
2022-07-26 09:05:48: at_reboot: stream.write(msg)
2022-07-26 09:05:48: at_reboot: UnicodeEncodeError: 'ascii' codec can't encode character '\u20ac' in position 93: ordinal not in range(128)
2022-07-26 09:05:48: at_reboot: Call stack:
2022-07-26 09:05:48: at_reboot: File "/var/www/html/openWB/runs/upgradeChargeLogs.py", line 27, in <module>
2022-07-26 09:05:48: at_reboot: log.info("upgrading charge logs with a price of " + str(args.price) + "\u20ac/kWh")
2022-07-26 09:05:48: at_reboot: Message: 'upgrading charge logs with a price of 0.3\u20ac/kWh'
```
Ich habe mich damit etwas mehr im Detail beschäftigt und auch eine Menge dazu rausgefunden. Mir ist aber nicht klar wie die optimale Lösung aussieht. Ich habe dazu eine [Frage auf Stackoverflow](https://stackoverflow.com/q/73118543/327301)

hhoefling, 2022-07-26:
Im Logging Umlaute und Sonderzeichen Vermeiden/umschreiben (z.b EUR)
Im GUI Php/Jscript das machen lassen.

yankee42, 2022-07-26:
> Im Logging Umlaute und Sonderzeichen Vermeiden/umschreiben (z.b EUR) Im GUI Php/Jscript das machen lassen.

Nein, das kann nicht die Lösung sein. Ja, in diesem einen Fall würde es gehen, weil man nur das `€` tauschen müsste, aber an anderen Stellen werden auch Eingaben von Nutzern geloggt und das Problem wird immer wieder kommen. Alles was durch das log geht zu filtern würde nicht nur die Logausgabe verzerren, sondern das Unterfangen würde auch größere Dimensionen annehmen.

Es geht ja nur um eine ungünstig gesetzte Umgebungsvariable. Die Ursache ist gefunden, jetzt braucht es nurnoch die bestmögliche Lösung. Das ist ein kleiner Schritt.

yankee42, 2022-08-14:
Um mal ein Update dazu zu geben was ich dazu recherchiert habe:
Die Locale wird für Apache in der `/etc/apache2/envvars` konfiguriert:
```
## The locale used by some modules like mod_dav
export LANG=C
## Uncomment the following line to use the system default locale instead:
#. /etc/default/locale
```
Um es zum rumtesten leicht zu machen. Auskommentieren geht schnell mit
```bash
sudo sed -i '/^\. \/etc\/default\/locale/{s/^/#/}' /etc/apache2/envvars
```
Einkommentieren mit:
```bash
sudo sed -i '/^#\. \/etc\/default\/locale/{s/^#//}' /etc/apache2/envvars
```

Das Problem ist: Damit die Änderung aktiv wird, muss der Apache neugestartet werden. Ein reload reicht nicht. Wenn man den Apache neustartet, dann werden dabei alle Prozesse getötet, die von PHP gestartet wurden. Das Verhalten lässt sich nicht durch `nohup` unterbinden. (Siehe dazu auch diese [Frage auf Stackoverflow](https://stackoverflow.com/questions/51706397)). D.h. alle Dienste die in der atreboot.sh gestartet werden sterben. Das betrifft aktuell `modbusserver.py`, `smarthomehandler.py`, `mqttsub.py` und `legacy_run_server.py`. Die Dienste werden dann vom cron5min neu gestartet, aber im schlimmstenfall steht für 5 Minuten alles still, was auf diesen Diensten beruht.

Ich habe noch keine befriedigende Lösung dafür. Es geht nicht um sonderlich viele Scripte, insofern wäre es möglich die `/etc/default/locale` zu beginn der betroffenen Scripte zu sourcen. Eine andere Variante wäre beim Update ein cronjob anzulegen, der einmalig läuft um sich danach selbst sofort wieder zu löschen. Alles recht kompliziert, nur um den Apache neuzustarten...

benderl, 2022-10-07:
Wenn ich mir [Deine Frage](https://stackoverflow.com/q/73118543/327301) ansehe, gibt es eine akzeptierte Lösung. Hast Du das schon getestet?

benderl, 2022-10-07:
In 2.0 ist das Update so aufgebaut, dass im Anschluss ein Reboot durchgeführt wird. Dadurch ersparen wir uns einige Abhängigkeiten. In 1.x wird es wohl vorerst so bleiben, wie es derzeit ist. Vielleicht kann über ein Flag in der Ramdisk signalisiert werden, dass ein Neustart erforderlich ist. Dann könnte dem Benutzer eine entsprechende Meldung angezeigt werden und der kann entscheiden, wann er das macht. Ist zwar für Linux nicht unbedingt typisch, aber relativ einfach umzusetzen.

hhoefling, 2022-10-07:
Ausserdem ist

`Das Problem ist: Damit die Änderung aktiv wird, muss der Apache neugestartet werden. Ein reload reicht nicht. Wenn man den Apache neustartet, dann werden dabei alle Prozesse getötet, die von PHP gestartet wurden. Das Verhalten lässt sich nicht durch nohup unterbinden. (Siehe dazu auch diese [Frage auf Stackoverflow](https://stackoverflow.com/questions/51706397)). D.h. alle Dienste die in der atreboot.sh gestartet werden sterben. Das betrifft aktuell modbusserver.py, smarthomehandler.py, mqttsub.py und legacy_run_server.py. `

Quatch.
Du kannst appache anhalten und alle diese Python Scripte laufen munter weiter.