snaptec/openWB #2346: Solarlog no more values

seaspotter, 2022-08-01:
Since Merge pull #2331 no more values are displayed for PV production, house consumption and grid import/export. Seems this bugfix is not working in current Nightly. Rollback to latest Beta 1.9.269 will fix the issue and everything is working like before

benderl, 2022-08-01:
Please post log messags from your openwb.log.

seaspotter, 2022-08-01:
Here the latest logs:

`2022-08-01 20:42:01: PID: 3575: **** Regulation loop start **** (LV1) at 70 main /var/www/html/openWB/regel.sh
graphing.sh: Zeile 11: 0.0: Syntaxfehler: Ungültiger arithmetischer Operator. (Fehlerverursachendes Zeichen ist \".0\").
2022-08-01 20:41:55: PID: 2211: Zeit zum abfragen aller Werte 4 Sekunden (LV1) at 176 main /var/www/html/openWB/regel.sh
2022-08-01 20:41:55: PID: 2211: chargestatlp1 0	chargestatlp2 0	chargestatlp3 0 (LV1) at 1304 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: plugstatlp1 0	plugstatlp2 0	plugstatlp3 0 (LV1) at 1303 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: lp1enabled 0	lp2enabled 0	lp3enabled 1 (LV1) at 1302 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: EVU 1:V/0A 2: V/0A 3: V/0A (LV1) at 1301 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: lla3 0	llv3 232.1	llas13 0 llas23  soclp1 0 soclp2 0 (LV1) at 1300 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: lla2 0	llv2 231.7	llas12 0 llas22  sofortll 16 hausverbrauch 0 wattbezug 0 uberschuss 0 (LV1) at 1299 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: lla1 0	llv1 229.1	llas11  llas21  mindestuberschuss 1600 abschaltuberschuss 200 lademodus 3 (LV1) at 1298 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: pv1watt 0 pv2watt  pvwatt 0 ladeleistung 0 llalt 0 nachtladen 0 nachtladen 0 minimalA 6 maximalA 16 (LV1) at 1297 loadvars loadvars.sh
2022-08-01 20:41:55: PID: 2211: etproviderprice  etprovidermaxprice  (LV1) at 1295 loadvars loadvars.sh
ValueError: invalid literal for int() with base 10: '554.0'
    watt2 = int(sys.argv[1])
  File "/var/www/html/openWB/runs/simcount.py", line 6, in <module>
Traceback (most recent call last):
2022-08-01 20:41:55: PID: 2211: ungültiger Wert für wattbezug: 554.0 (LV0) at 929 loadvars loadvars.sh
loadvars.sh: Zeile 663: [: -ge: Einstelliger (unärer) Operator erwartet.
2022-08-01 20:41:51: PID: 2211: **** Regulation loop start **** (LV1) at 70 main /var/www/html/openWB/regel.sh
2022-08-01 20:41:41: PID: 2117: **** Regulation loop start **** (LV1) at 70 main /var/www/html/openWB/regel.sh
graphing.sh: Zeile 11: 0.0: Syntaxfehler: Ungültiger arithmetischer Operator. (Fehlerverursachendes Zeichen ist \".0\").
2022-08-01 20:41:36: PID: 753: Zeit zum abfragen aller Werte 5 Sekunden (LV1) at 176 main /var/www/html/openWB/regel.sh
2022-08-01 20:41:35: PID: 753: chargestatlp1 0	chargestatlp2 0	chargestatlp3 0 (LV1) at 1304 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: plugstatlp1 0	plugstatlp2 0	plugstatlp3 0 (LV1) at 1303 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: lp1enabled 0	lp2enabled 0	lp3enabled 1 (LV1) at 1302 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: EVU 1:V/0A 2: V/0A 3: V/0A (LV1) at 1301 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: lla3 0	llv3 232.3	llas13 0 llas23  soclp1 0 soclp2 0 (LV1) at 1300 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: lla2 0	llv2 231.8	llas12 0 llas22  sofortll 16 hausverbrauch 0 wattbezug 0 uberschuss 0 (LV1) at 1299 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: lla1 0	llv1 229.4	llas11  llas21  mindestuberschuss 1600 abschaltuberschuss 200 lademodus 3 (LV1) at 1298 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: pv1watt 0 pv2watt  pvwatt 0 ladeleistung 0 llalt 0 nachtladen 0 nachtladen 0 minimalA 6 maximalA 16 (LV1) at 1297 loadvars loadvars.sh
2022-08-01 20:41:35: PID: 753: etproviderprice  etprovidermaxprice  (LV1) at 1295 loadvars loadvars.sh
ValueError: invalid literal for int() with base 10: '558.0'
    watt2 = int(sys.argv[1])
  File "/var/www/html/openWB/runs/simcount.py", line 6, in <module>
Traceback (most recent call last):
2022-08-01 20:41:35: PID: 753: ungültiger Wert für wattbezug: 558.0 (LV0) at 929 loadvars loadvars.sh
loadvars.sh: Zeile 663: [: -ge: Einstelliger (unärer) Operator erwartet.
2022-08-01 20:41:31: PID: 753: **** Regulation loop start **** (LV1) at 70 main /var/www/html/openWB/regel.sh
2022-08-01 20:41:21: PID: 646: **** Regulation loop start **** (LV1) at 70 main /var/www/html/openWB/regel.sh
graphing.sh: Zeile 11: 0.0: Syntaxfehler: Ungültiger arithmetischer Operator. (Fehlerverursachendes Zeichen ist \".0\").
2022-08-01 20:41:16: PID: 31771: Zeit zum abfragen aller Werte 5 Sekunden (LV1) at 176 main /var/www/html/openWB/regel.sh
2022-08-01 20:41:15: PID: 31771: chargestatlp1 0	chargestatlp2 0	chargestatlp3 0 (LV1) at 1304 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: plugstatlp1 0	plugstatlp2 0	plugstatlp3 0 (LV1) at 1303 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: lp1enabled 0	lp2enabled 0	lp3enabled 1 (LV1) at 1302 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: EVU 1:V/0A 2: V/0A 3: V/0A (LV1) at 1301 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: lla3 0	llv3 232.5	llas13 0 llas23  soclp1 0 soclp2 0 (LV1) at 1300 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: lla2 0	llv2 231.7	llas12 0 llas22  sofortll 16 hausverbrauch 0 wattbezug 0 uberschuss 0 (LV1) at 1299 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: lla1 0	llv1 230.1	llas11  llas21  mindestuberschuss 1600 abschaltuberschuss 200 lademodus 3 (LV1) at 1298 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: pv1watt 0 pv2watt  pvwatt 0 ladeleistung 0 llalt 0 nachtladen 0 nachtladen 0 minimalA 6 maximalA 16 (LV1) at 1297 loadvars loadvars.sh
2022-08-01 20:41:15: PID: 31771: etproviderprice  etprovidermaxprice  (LV1) at 1295 loadvars loadvars.sh
ValueError: invalid literal for int() with base 10: '551.0'
    watt2 = int(sys.argv[1])
  File "/var/www/html/openWB/runs/simcount.py", line 6, in <module>
Traceback (most recent call last):
2022-08-01 20:41:14: PID: 31771: ungültiger Wert für wattbezug: 551.0 (LV0) at 929 loadvars loadvars.sh
loadvars.sh: Zeile 663: [: -ge: Einstelliger (unärer) Operator erwartet.`

benderl, 2022-08-01:
There is no message from the solarlog module. Please use debug mode 2. And also please take care formatting the log messages here. The ones above are not really easy to read.

seaspotter, 2022-08-01:
Sorry hope it works now:

hausverbrauch, wattbezug, uberschuss  are all always zero, if I change back to .169 it is working perfectly and hausverbrauch and wattbezug have proper values. Tried that now at least 5-6 times with down/upgrade

Replaced Solarlog IP in the log below with "Sol.arl.og.IP":

`2022-08-01 21:14:51: PID: 29329: **** Regulation loop needs 0 seconds (LV2) at 55 cleanup /var/www/html/openWB/regel.sh
2022-08-01 21:14:51: PID: 29329: **** Regulation loop start **** (LV1) at 70 main /var/www/html/openWB/regel.sh
2022-08-01 21:14:49: PID: 28014: **** WARNING **** Regulation loop needs 8 seconds (LV0) at 57 cleanup /var/www/html/openWB/regel.sh
graphing.sh: Zeile 11: 0.0: Syntaxfehler: Ungültiger arithmetischer Operator. (Fehlerverursachendes Zeichen ist \".0\").
2022-08-01 21:14:48: PID: 28014: Zeit zum abfragen aller Werte 7 Sekunden (LV1) at 176 main /var/www/html/openWB/regel.sh
2022-08-01 21:14:48: PID: 28014: chargestatlp1 0	chargestatlp2 0	chargestatlp3 0 (LV1) at 1304 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: plugstatlp1 0	plugstatlp2 0	plugstatlp3 0 (LV1) at 1303 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: lp1enabled 0	lp2enabled 0	lp3enabled 1 (LV1) at 1302 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: EVU 1:V/0A 2: V/0A 3: V/0A (LV1) at 1301 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: lla3 0	llv3 231.7	llas13 0 llas23  soclp1 0 soclp2 0 (LV1) at 1300 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: lla2 0	llv2 233.0	llas12 0 llas22  sofortll 16 hausverbrauch 0 wattbezug 0 uberschuss 0 (LV1) at 1299 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: lla1 0	llv1 229.9	llas11  llas21  mindestuberschuss 1600 abschaltuberschuss 200 lademodus 3 (LV1) at 1298 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: pv1watt 0 pv2watt  pvwatt 0 ladeleistung 0 llalt 0 nachtladen 0 nachtladen 0 minimalA 6 maximalA 16 (LV1) at 1297 loadvars loadvars.sh
2022-08-01 21:14:48: PID: 28014: etproviderprice  etprovidermaxprice  (LV1) at 1295 loadvars loadvars.sh
ValueError: invalid literal for int() with base 10: '555.0'
    watt2 = int(sys.argv[1])
  File "/var/www/html/openWB/runs/simcount.py", line 6, in <module>
Traceback (most recent call last):
2022-08-01 21:14:47: PID: 28014: ungültiger Wert für wattbezug: 555.0 (LV0) at 929 loadvars loadvars.sh
2022-08-01 21:14:47: PID: 28406: RET: 0 (LV2) at 19 main modules/bezug_solarlog/main.sh
2022-08-01 21:14:47: PID: 19824: legacy run server: Completed running command in 0.09s: ["bezug_solarlog.solarlog","Sol.arl.og.IP","0"]
2022-08-01 21:14:47: PID: 19824: Solarlog EVU: PV Leistung: 0.0
2022-08-01 21:14:47: PID: 19824: Solarlog EVU: Leistung: 555.0
2022-08-01 21:14:47: PID: 19824: requests.packages.urllib3.connectionpool: http://Sol.arl.og.IP:80 "POST /getjp HTTP/1.1" 200 None
2022-08-01 21:14:47: PID: 19824: requests.packages.urllib3.connectionpool: Starting new HTTP connection (1): Sol.arl.og.IP:80
2022-08-01 21:14:47: PID: 19824: Solarlog EVU: Solarlog Speicher: 0
2022-08-01 21:14:47: PID: 19824: Solarlog EVU: Solarlog IP: Sol.arl.og.IP
2022-08-01 21:14:47: PID: 19824: legacy run server: Received command ["bezug_solarlog.solarlog","Sol.arl.og.IP","0"]
2022-08-01 21:14:47: PID: 28406: Bezug Solarlog Speicher : 0 (LV2) at 14 main modules/bezug_solarlog/main.sh
2022-08-01 21:14:47: PID: 28406: Bezug Solarlog IP: Sol.arl.og.IP (LV2) at 13 main modules/bezug_solarlog/main.sh
loadvars.sh: Zeile 663: [: -ge: Einstelliger (unärer) Operator erwartet.
2022-08-01 21:14:46: PID: 28014: Calling SoC-Module: modules/soc_http/main.sh for chargepoint 2 (LV2) at 23 run_soc_module loadvars.sh
2022-08-01 21:14:46: PID: 28014: Request to run SoC-Module: soc_httplp2 (LV2) at 4 run_soc_module loadvars.sh
requests.exceptions.ReadTimeout: HTTPConnectionPool(host='Sol.arl.og.IP', port=80): Read timed out. (read timeout=3)
raise ReadTimeout(e, request=request)
2022-08-01 21:14:45: PID: 28014: ungültiger Wert für pvwatt: 0.0 (LV0) at 490 loadvars loadvars.sh
File "/usr/lib/python3/dist-packages/requests/adapters.py", line 499, in send
r = adapter.send(request, **kwargs)
File "/usr/lib/python3/dist-packages/requests/sessions.py", line 609, in send
resp = self.send(prep, **send_kwargs)
File "/usr/lib/python3/dist-packages/requests/sessions.py", line 488, in request
return session.request(method=method, url=url, **kwargs)
File "/usr/lib/python3/dist-packages/requests/api.py", line 56, in request
return request('post', url, data=data, json=json, **kwargs)
File "/usr/lib/python3/dist-packages/requests/api.py", line 110, in post
response = requests.post("http://"+bezug_solarlog_ip+'/getjp', data=data, timeout=3).json()
File "/var/www/html/openWB/modules/wr_solarlog/solarlog.py", line 19, in update
parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
args.RUN(args)
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 34, in run_using_positional_cli_args
run_using_positional_cli_args(update, argv)
File "/var/www/html/openWB/modules/wr_solarlog/solarlog.py", line 33, in main
importlib.import_module(parsed[0]).main(parsed[1:])
File "/var/www/html/openWB/packages/legacy_run_server.py", line 109, in handle_message
self.__callback(read_all_bytes(connection))
[... cut ...]
2022-08-01 21:14:24: PID: 19824: requests.packages.urllib3.connectionpool: http://Sol.arl.og.IP:80 "POST /getjp HTTP/1.1" 200 None
2022-08-01 21:14:24: PID: 19824: requests.packages.urllib3.connectionpool: Starting new HTTP connection (1): Sol.arl.og.IP:80
2022-08-01 21:14:24: PID: 19824: Solarlog EVU: Solarlog Speicher: 0
2022-08-01 21:14:24: PID: 19824: Solarlog EVU: Solarlog IP: Sol.arl.og.IP
2022-08-01 21:14:24: PID: 19824: legacy run server: Received command ["bezug_solarlog.solarlog","Sol.arl.og.IP","0"]`

seaspotter, 2022-08-03:
@LKuemmel any idea? Because commit was from you regarding the change in Solarlog and since then it is not working (at least for me)

benderl, 2022-08-11:
#2356

closed 2022-08-11 by benderl