snaptec/openWB #2350: at_reboot.sh Prozess überlebt den Updatevorgang dauerhaft

yankee42, 2022-08-09:
Zu Dokumentationszwecken und potentiell zur Diskussion:

@DetMoerk hat [hier](https://github.com/snaptec/openWB/commit/001404d5784ad8b979907101ee8ee28f840ab8f7#commitcomment-80630555) den Kommentar geschrieben:

> Seit dem Umbau auf openwbRunLoggingOutput, bleibt ein atreboot.sh Prozess auch nach dem 10 Minuten sleep am Laufen. der strace hängt in einem read, daher vermute ich, dass er die read Schleife in der Funktion nie verlässt.

Ich hatte auch zunächst keine Ahnung warum das so ist. Tatsächlich habe ich mehrere Stunden gebraucht bis ich dahinter gekommen bin. Um das ganze zu dokumentieren, falls in Zukunft nochmal jemand darüber stolpert schreibe ich die Erkentnisse hier auf.

Wir haben mehrere Dienste die in der at_reboot.sh gestartet werden. Zum BEISPIEL:

https://github.com/snaptec/openWB/blob/4836d995dd0a4af8d20e52300560614761190d30/runs/atreboot.sh#L92
https://github.com/snaptec/openWB/blob/4836d995dd0a4af8d20e52300560614761190d30/runs/atreboot.sh#L127

Wenn ich die gestarten Programm beende, dann enden auch die verbleibenden `at_reboot.sh`-Prozesse:
```
$ ps aux | grep atreboot; sudo pkill -f 'modbusserver.py|smarthomehandler.py|mqttsub.py|legacy_run_server.py'; sleep 1; echo "Remaining after kill:"; ps aux | grep atreboot
pi         512  0.1  0.3   5308  3336 ?        S    08:32   0:00 /bin/bash /var/www/html/openWB/runs/atreboot.sh
pi         523  0.0  0.3   5308  2916 ?        S    08:32   0:00 /bin/bash /var/www/html/openWB/runs/atreboot.sh
pi        3475  0.0  0.0   4368   544 pts/0    S+   08:33   0:00 grep --color=auto atreboot
Remaining after kill:
pi        3613  0.0  0.0   4368   560 pts/0    S+   08:33   0:00 grep --color=auto atreboot
```
Das `read` bleibt weiterhin aktiv und würde es ins Log schreiben, wenn von den entsprechenden Programmen eine Ausgabe kommt. Ich würde sagen "it's not a bug - it's a feature". Denn auf diese Weise geht potentieller Output dieser Prozesse, der in Fehlerfällen zum Debuggen nützlich wäre nicht verloren.

Ein Nachteil an dem Verhalten fällt mir nicht ein.

hhoefling, 2022-08-09:
Die gestarteten Daemons werden alle mit & in den Huntergrund geschickt und haben keine Verbindung mehr zum STDOUT des Scripts. Also kommt bei "read"  auch sicher nichts mehr an wenn die letzte Zeile von at_reboot.sh  ausgeführt wurde.

Ich selbst habe die Umleitung der Ausgabe noch im Aufrufer (crontab etc) stehen (also das >>../openWB.log)
Da bleibt nix hängen. Ich denke die ungewöhnliche Konstruction mit openwbRunLoggingOutput ist der Auslöser des hängers.

DetMoerk, 2022-08-09:
Wäre der normale Weg mit openwbDebugLog hier nicht die bessere Lösung, nur um die Ausgaben im Log aufzuhübschen?

Alternative: kann nohup in Scripten verwendet weden, dann sind die Prozesse wirklich vom aufrufenden Programm abgehängt.

VG
Det

yankee42, 2022-08-09:
> Die gestarteten Daemons werden alle mit & in den Huntergrund geschickt und haben keine Verbindung mehr zum STDOUT des Scripts. Also kommt bei "read" auch sicher nichts mehr an wenn die letzte Zeile von at_reboot.sh ausgeführt wurde.

Ähm... Doch? Nur weil die asynchron laufen ist doch STDOUT nicht weg.

> Ich selbst habe die Umleitung der Ausgabe noch im Aufrufer (crontab etc) stehen (also das >>../openWB.log) Da bleibt nix hängen.

Wenn nicht auf stdout ausgegeben wird kommt natürlich auch nichts.

> Ich denke die ungewöhnliche Konstruction mit openwbRunLoggingOutput ist der Auslöser des hängers.

"Hänger" klingt so als wäre irgendwas nicht in Ordnung. Aber es funktioniert alles so wie es soll.

> Wäre der normale Weg mit openwbDebugLog hier nicht die bessere Lösung, nur um die Ausgaben im Log aufzuhübschen?

Nein. Das ist deutlich mehr Aufwand. Du musst jede einzelne echo-Zeile dafür anfassen. Es ist auch fehleranfällig, denn leicht geht dabei was durch. Das Log dient ja gerade dazu bei der Fehlersuche zu helfen, wenn etwas nicht funktioniert und dann kommen möglicherweise von Befehlen Ausgaben mit denen man nicht gerechnet hat. Jeder einzelne Befehl müsste dann in ein solches Logstatement gebracht werden, damit nichts untergeht. Und die Frage ist dann auch: Wofür? Nur damit die Ausgabe von `ps` eine Zeile weniger hat?? Am aktuellen Verhalten ist nichts kaputt.

> Alternative: kann nohup in Scripten verwendet weden, dann sind die Prozesse wirklich vom aufrufenden Programm abgehängt. Habe dazu disown gefunden, aber meine ersten Tests haben keine Besserung ergeben. atreboot bleibt trotzdem dauerhaft aktiv :-(.

Du kannst auch jeden einzelnen Befehl zu /dev/null pipen. Aber das ist wider der ganzen Intention, dass potentielle Fehler im Log landen. Warum willst du denn unbedingt, dass `atreboot` endet?

hhoefling, 2022-08-09:
> Du kannst auch jeden einzelnen Befehl zu /dev/null pipen. Aber das ist wider der ganzen Intention, dass potentielle Fehler im Log landen. Warum willst du denn unbedingt, dass `atreboot` endet?

Damit die Zeilen
  	    echo 0 >/var/www/html/openWB/ramdisk/bootinprogress;  
	    echo 0 >/var/www/html/openWB/ramdisk/updateinprogress
auch zum Zuge kommen (nach 600 Sekunden "hängen")

yankee42, 2022-08-09:
> Damit die Zeilen echo 0 >/var/www/html/openWB/ramdisk/bootinprogress; echo 0 >/var/www/html/openWB/ramdisk/updateinprogress auch zum Zuge kommen (nach 600 Sekunden "hängen")

Die kommen auch jetzt zum Zuge.

hhoefling, 2022-08-09:
Klar normales Ende des Scripts, 
dann ist der Watchdog überflüssig.
Wenn aber z.b git hängt (Internet problem) wird keiner updateinprogress auflösen.
Genau das sollte diese Zeile aber nach 600 Sekunden sicherstellen damit die GUI wieder aktiviert wird.
Wenn das nun "hängt"  bis du sicher das die Zeilen (nach 600 Sekuden) durchlaufen wurden?

DetMoerk, 2022-08-09:
`Nein. Das ist deutlich mehr Aufwand. Du musst jede einzelne echo-Zeile dafür anfassen. Es ist auch fehleranfällig, denn leicht geht dabei was durch. Das Log dient ja gerade dazu bei der Fehlersuche zu helfen, wenn etwas nicht funktioniert und dann kommen möglicherweise von Befehlen Ausgaben mit denen man nicht gerechnet hat. Jeder einzelne Befehl müsste dann in ein solches Logstatement gebracht werden, damit nichts untergeht. Und die Frage ist dann auch: Wofür? Nur damit die Ausgabe von ps eine Zeile weniger hat?? Am aktuellen Verhalten ist nichts kaputt.`

Auf der einen Seite bewundere ich deine Fähigkeiten, Dinge zu machen, aber ab und zu ist KISS auch eine bessere Lösung. Die Funktion openWBDebugLog gibt es bereits und wird in allen anderen Modulen, die noch unter der Bash laufen auch genutzt. Einmal über die atreboot drüber gehen, die Echos in openwbdebuglog ändern und es wäre erledigt, wenn einem die bisherigen Ausgaben ohne Zeitstempel nicht gefallen. 
Ich sehe das auch als Hänger, wenn es einen immer noch laufenden atreboot Prozess gibt, der nicht beendet wird, obwohl alles abgeschlossen ist! Du hast selbst gesagt, dass du Stunden gesucht hast, warum das nicht beendet wird. Sorry, aber die Überarbeitung mit openWBDebugLog wäre sicher weniger Zeitaufwand gewesen und zudem auch für die Leute lesbar, die nicht auf deinem Niveau programmieren.

yankee42, 2022-08-09:
> Auf der einen Seite bewundere ich deine Fähigkeiten, Dinge zu machen, aber ab und zu ist KISS auch eine bessere Lösung.

KISS ist schonmal besser. Eben deswegen ja. Die meisten Leute die an diesem Skript arbeiten werden innerhalb der at_reboot-Funktion bleiben. Und da sind sie gut aufgehoben, man kann da nicht wirklich viel falsch machen. Alles was ausgegeben wird, egal ob es eine Fehlermeldung von einem fehlgeschlagenem `cp` ist oder ein `echo` landet mit Name und Timestamp im Log. Das ist doch wirklich einfacher, als die `openWBDebugLog`.

Wer dann über sich hinaus wächst und etwas an "meiner" `openwbRunLoggingOutput` machen möchte, muss etwas mehr über Bash wissen. Aber auch nicht wirklich viel. Die Funktion hat gerade mal 7 Zeilen und das sind weder besonders lange, noch auf andere Art und Weise komplexe Zeilen.

> Die Funktion openWBDebugLog gibt es bereits und wird in allen anderen Modulen, die noch unter der Bash laufen auch genutzt. Einmal über die atreboot drüber gehen, die Echos in openwbdebuglog ändern und es wäre erledigt, wenn einem die bisherigen Ausgaben ohne Zeitstempel nicht gefallen. 

Genau WAS wäre damit erledigt? Herauszufinden warum die `at_reboot.sh` "hängen" bleibt? Nope. Schlauer bin ich dadurch nicht.

Überall `openWBDebugLog` einzufügen würde an der Stelle Zeitstempel einfügen, wo auch genau der Befehl steht. Aber nicht bei Ausgaben von anderen Kommandos. Und wir wollen garnicht eine Ausgabe von jedem Kommando, sofern kein Fehler vorliegt. Eine Lösung könnte so aussehen:
```bash
X=$(cp foo bar 2>&1); [ -z "$X" ] || openWBDebugLog "MAIN" 0 "atreboot.sh: Could not copy..."
```
Und das dann eben für jede einzelne Zeile. Dann sieht man aber den Code vor Overhead nicht mehr. Das macht hoffentlich keiner. So wie es jetzt ist, ist es nur ein kleiner Funktionsaufruf, der alles mit wenig Codeeinsatz abarbeitet.

> Ich sehe das auch als Hänger, wenn es einen immer noch laufenden atreboot Prozess gibt, der nicht beendet wird, obwohl alles abgeschlossen ist!

Das ist bedauerlich, dass du das so siehst, das hat operationell auf die openWB aber keine Auswirkungen.

> Du hast selbst gesagt, dass du Stunden gesucht hast, warum das nicht beendet wird.

Richtig. Weil ich wissen wollte warum. Man hätte es auch einfach lassen können, denn es liegt kein Fehlverhalten vor. Dennoch interessiert mich was die von mir geschriebene Software tut und kann.

> Sorry, aber die Überarbeitung mit openWBDebugLog wäre sicher weniger Zeitaufwand gewesen und zudem auch für die Leute lesbar, die nicht auf deinem Niveau programmieren.

Stimmt, weniger Zeitaufwand wäre es gewesen. Und weniger zu lernen. Man hätte dem freiwilligen Support im Forum und den professionellen Support bei openWB das Leben weniger leicht gemacht. Neue Entwickler die jetzt nur etwas innerhalb der Funktion at_reboot ändern müssen, brauchen diese ohnehin nicht zu verstehen. Der Code so ist robuster für was Leute auch immer damit anstellen. Das alles aufgeben, dafür, dass ich an dem einen Problem schneller einen Haken machen kann? Ein Problem welches nach etwas Recherche nichtmal überhaupt ein Problem ist und keine Handlung bedarf?

DetMoerk, 2022-08-09:
> Das ist bedauerlich, dass du das so siehst, das hat operationell auf die openWB aber keine Auswirkungen.

Da sind wir wohl komplett anderer Ansicht. Ich will keine Prozesse sehen, die nur noch nutzlos rumhängen. Auch wenn es der openWB Software so wohl nicht schadet, frisst auch ein solcher Prozess Speicher und andere Ressourcen. Würdest du es akzeptieren, dass jedes Programm in so einem Zustand einfach stehen bleibt? 
Kleinvieh macht dann auch irgendwann so viel Mist, dass es auffällt oder weh tut. 

Schön, dass es jetzt formatierte Ausgaben im openWB.log gibt, aber lieber sehe ich unformatierte Ausgaben OHNE Seiteneffekte.

Nur meine Meinung zu dem Thema.

hhoefling, 2022-08-09:
Ich halte den Häger auch für ein Fehler.
Vor allem wo es vorher mal (keine Ahnung bis wann) nicht so war.

Gerade wenn man testweise mehrer Update's fährt (ok machen nur wir) 
dann wird's bei ps langsam unübersichtlich.

Also bitte mal testen ob die Freigabelogig trotz hänger überhaupt noch arbeitet.

Z..b mal die 600 mal auf 100 runtersetzen und kurz vor ende des atreboot ein "sleep 120" reinsetzen
Dann müste nach 100 Sekunden der "kill" kommen und die Gui wieder freigeben.
Anschliessen sollte KEINE atreboot.sh mehr da sein und die beiden flags gelöscht sein.

benderl, 2022-08-11:
Der Abbruch nach 600 Sekunden funktioniert weiterhin, das ist nicht das Problem.
Dass ein Prozess übrig bleibt, der vom Namen mit dem Booten zusammenhängt, ist Geschmackssache. So richtig gefällt mir das auch nicht. Ich teste gerade eine Variante, welche die gestarteten Prozesse mit `nohup` komplett von STDOUT trennt.

DetMoerk, 2022-08-11:
Eine Sache fällt mir dazu auch noch ein. Eigentlich kenne ich solche Prozesse, die hier asynchron angestoßen werden als systemd services. Damit sind dann solche Nebeneffekte weg, dass der eine Prozess mit sudo, der andere ohne sudo gestartet wird... 
Das jetzt in die 1.9 einzubauen wäre wohl ein etwas größerer Aufwand und wäre der aktuell eh etwas schwankenden Stabilität sicher nicht zuträglich. Aber für die V2.0 die ja bereits selbst als Service gestartet wird, sollte man das evtl. berücksichtigen.

DetMoerk, 2022-08-12:
> Der Abbruch nach 600 Sekunden funktioniert weiterhin, das ist nicht das Problem. Dass ein Prozess übrig bleibt, der vom Namen mit dem Booten zusammenhängt, ist Geschmackssache. So richtig gefällt [...]

atreboot.sh Prozess direkt nach einen Update: 
```pi@raspi4:/var/www/html/openWB/ramdisk $ ps -ef |grep atre
pi        4259  3977  0 15:37 pts/0    00:00:00 grep --color=auto atre
pi        7461     1  0 15:28 ?        00:00:00 /bin/bash /var/www/html/openWB/runs/atreboot.sh
pi        7467  7461  0 15:28 ?        00:00:00 /bin/bash /var/www/html/openWB/runs/atreboot.sh
pi        7468     1  0 15:28 ?        00:00:00 /bin/bash /var/www/html/openWB/runs/atreboot.sh
```

Bei mir bleibt leider immer noch der letzte atreboot.sh Prozess offen nach 10 Minuten:
```pi@raspi4:/var/www/html/openWB/ramdisk $ ps -ef |grep atre
pi        7467     1  0 15:28 ?        00:00:00 /bin/bash /var/www/html/openWB/runs/atreboot.sh
pi       15276  3977  0 15:39 pts/0    00:00:00 grep --color=auto atre
```
Und der Prozess hat noch die Pipe auf STDIN offen und die STDOUT und STDERR auf dem openWB.log 
```
pi@raspi4:/var/www/html/openWB/ramdisk $ sudo lsof -p 7467
COMMAND    PID USER   FD   TYPE DEVICE SIZE/OFF    NODE NAME
atreboot. 7467   pi  cwd    DIR    8,2     4096 1443042 /var/www/html
atreboot. 7467   pi  rtd    DIR    8,2     4096       2 /
atreboot. 7467   pi  txt    REG    8,2   925124 1703951 /bin/bash
atreboot. 7467   pi  mem    REG    8,2  1291908 3941031 /lib/arm-linux-gnueabihf/libc-2.28.so
atreboot. 7467   pi  mem    REG    8,2     9768 3941034 /lib/arm-linux-gnueabihf/libdl-2.28.so
atreboot. 7467   pi  mem    REG    8,2   133216 3932172 /lib/arm-linux-gnueabihf/libtinfo.so.6.1
atreboot. 7467   pi  mem    REG    8,2    17708 2502812 /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
atreboot. 7467   pi  mem    REG    8,2   138608 3941017 /lib/arm-linux-gnueabihf/ld-2.28.so
atreboot. 7467   pi    0r  FIFO   0,11      0t0 1357752 pipe
atreboot. 7467   pi    1w   REG    8,2   186524 5636566 /var/log/openWB.log
atreboot. 7467   pi    2w   REG    8,2   186524 5636566 /var/log/openWB.log
pi@raspi4:/var/www/html/openWB/ramdisk $
```
Hat leider nicht so geholfen, wie gehofft.

```
pi@raspi4:/var/www/html/openWB/ramdisk $ sudo lsof -u pi |grep 1357752
atreboot.  7467   pi    0r     FIFO       0,11      0t0    1357752 pipe
bash       8273   pi    1w     FIFO       0,11      0t0    1357752 pipe
bash       8273   pi    2w     FIFO       0,11      0t0    1357752 pipe
bash       8276   pi    1w     FIFO       0,11      0t0    1357752 pipe
bash       8276   pi    2w     FIFO       0,11      0t0    1357752 pipe
pi@raspi4:/var/www/html/openWB/ramdisk $
```
der legacy_run_server scheint noch an der Pipe zu nuckeln.
``` 
pi@raspi4:/var/www/html/openWB/ramdisk $ ps -ef |grep  "8273\|8276"
pi        6694  3977  0 16:01 pts/0    00:00:00 grep --color=auto 8273\|8276
pi        8273     1  0 15:28 ?        00:00:00 bash /var/www/html/openWB/packages/legacy_run_server.sh
root      8275  8273  0 15:28 ?        00:00:00 sudo -u pi python3 /var/www/html/openWB/packages/legacy_run_server.py
pi        8276  8273  0 15:28 ?        00:00:00 bash /var/www/html/openWB/packages/legacy_run_server.sh
pi@raspi4:/var/www/html/openWB/ramdisk $
```

hhoefling, 2022-08-12:
Ja der legacy_run_server hat wohl doch noch verspätet nebenwirkungen.
Ich benutze ihn nicht (Verwende die 244 RCT python module direkt) und  habe den start auskommentiert.
Daher bei mir keine "atreboot"-Leichen, :-)

hhoefling, 2022-08-12:
Hm....
richtig das da zwei sind?

```
pi@raspi4:/var/www/html/openWB/ramdisk $ ps -ef |grep  "8273\|8276"
pi        6694  3977  0 16:01 pts/0    00:00:00 grep --color=auto 8273\|8276
pi        8273     1  0 15:28 ?        00:00:00 bash /var/www/html/openWB/packages/legacy_run_server.sh
root      8275  8273  0 15:28 ?        00:00:00 sudo -u pi python3 /var/www/html/openWB/packages/legacy_run_server.py
pi        8276  8273  0 15:28 ?        00:00:00 bash /var/www/html/openWB/packages/legacy_run_server.sh
pi@raspi4:/var/www/html/openWB/ramdisk $
```

DetMoerk, 2022-08-12:
Der LRS wird erst im legacy_run_server.sh selbst in den Hintergrund geschickt. 
```
pi@raspi4:/var/www/html/openWB $ grep -R "legacy_run_server.sh"
Übereinstimmungen in Binärdatei .git/index
grep: packages/legacy_run_server.sock: Kein passendes Gerät bzw. keine passende Adresse gefunden
runs/atreboot.sh.det:   bash "$OPENWBBASEDIR/packages/legacy_run_server.sh"
runs/atreboot.sh.org:   bash "$OPENWBBASEDIR/packages/legacy_run_server.sh"
runs/atreboot.sh:       bash "$OPENWBBASEDIR/packages/legacy_run_server.sh"
runs/cron5min.sh:       bash "$OPENWBBASEDIR/packages/legacy_run_server.sh"
pi@raspi4:/var/www/html/openWB $
```
Wird an 2 Stellen aufgerufen. cron5min.sh und atreboot.sh beide als pi User. Damit gibt es ein Problem, dass der erste pkill bleibt am LRS.sh hängen, das root gehört. Das hat Yankee an anderer Stelle schon beispielhaft gelöst. pkill -u pi ...

Wenn der LRS einmal neu gestartet wird, ist auch der atreboot dann endlich zufrieden und und beendet sich.

DetMoerk, 2022-08-12:
@benderl: Wenn ich das jetzt richtig sehe, wird der LRS nur von atreboot und cron5min gestartet. Beide als user pi. 
braucht es dann eigentlich den Umweg, den LRS über sudo -u pi quasi unter root dann wieder als pi zu starten?
https://github.com/snaptec/openWB/blob/130dca32cfb5524e1fa816f2ff5eb9b4acab6e81/packages/legacy_run_server.sh#L41
Das sollte sich doch jetzt auch überholt haben. 
Ich teste das gerade bei mir mal, den sudo weg zu lassen.

benderl, 2022-08-12:
@DetMoerk 
Machst Du einen PR mit den Anpassungen?

DetMoerk, 2022-08-12:
@benderl: mir ist gerade erst aufgefallen, dass auch das LRS Start Script von @yankee42 ist. 
Dort auch diese Konstruktion mit dem Endlos Loop für die Log Ausgaben drin. Ich hab es jetzt nicht nicht geschafft, das mit nohup abzuhängen. Evtl. schaue ich mir das morgen noch mal an.

DetMoerk, 2022-08-12:
> Hm....
> richtig das da zwei sind?

Das kommt auch durch den while read für die Umleitung der einzelnen STDOUT Zeilen in das Logfile.
Warum man das an der Stelle so braucht verstehe ich leider nicht.

hhoefling, 2022-08-12:
```
pi@raspi4:/var/www/html/openWB $ ps -ef |grep atr
pi         491     1  0 17:28 ?        00:00:00 /bin/bash /var/www/html/openWB/runs/atreboot.sh
pi       22783   944  0 17:40 pts/1    00:00:00 grep --color=auto atr
pi@raspi4:/var/www/html/openWB $ ./packages/legacy_run_server.sh
Server is already running. Killing old instance.
Old process successfully terminated.
Starting legacy run server
pi@raspi4:/var/www/html/openWB $ ps -ef |grep atr
pi       25523   944  0 17:41 pts/1    00:00:00 grep --color=auto atr
pi@raspi4:/var/www/html/openWB $

```
Mh... Sieht auch komisch aus
Erst  kommt "Starting legacy run server"
und dann ist da keine :-(

yankee42, 2022-08-13:
> @benderl: Wenn ich das jetzt richtig sehe, wird der LRS nur von atreboot und cron5min gestartet. Beide als user pi. braucht es dann eigentlich den Umweg, den LRS über sudo -u pi quasi unter root dann wieder als pi zu starten?

`sudo -u pi` ist nicht "unter root wieder als pi", sondern einfach nur "als pi".

Aber nein, das wird nicht (mehr) benötigt. Als der legacy-run-server neu war, wurde atreboot noch als root ausgeführt. Da das nicht mehr der Fall ist könnte das sudo an der Stelle auch weg. Wenn es da bleibt schadet es allerdings auch nicht, immerhin dokumentiert es so gut sichtbar, dass das Programm immer als pi laufen muss, ohne dass man alle Referenzen durchsuchen muss. Ob das hilfreich ist...

> @hhoefling
> 
> > Hm....
> > richtig das da zwei sind?
> 
> Das kommt auch durch den while read für die Umleitung der einzelnen STDOUT Zeilen in das Logfile. Warum man das an der Stelle so braucht verstehe ich leider nicht.

Der Grund warum ich das eingebaut habe ist weil ich dachte bash würde hier einen fortlaufenden File-Handle öffnen. In dem Fall würden sich mehrere Prozesse die gleichzeitig ins Log schreiben gegenseitig ihre Ausgaben überschreiben. Ich hatte Hinweise gefunden die darauf hindeuteten, dass dies passiert ist. Das war aber offenbar eine Fehlinterpretation, [sagt zumindest jemand im Internet](https://stackoverflow.com/a/598483059). Geprüft habe ich das zwar nicht, aber wahrscheinlich kann das ganze read durch ein einfaches redirect to file ersetzt werden.

hhoefling, 2022-08-13:
Ich würde das Einzelzeilenlogging openWBDebugLog
Bevorzugen. (Auch in den subscripten)
In der Zeit die wir uns damit beschäftigen hätte ich die 20 Echos schon durch.

DetMoerk, 2022-08-14:
> `sudo -u pi` ist nicht "unter root wieder als pi", sondern einfach nur "als pi".

Du solltest dir mal oben in meiner Antwort die Prozessliste anschauen. Evtl. fällt dir das dann auch was auf! Du bist selbst schon 2 Mal darüber gefallen, dass es einen zugehörigen root Prozess gibt. Bei den pkills, die du dann mit -u pi eingeschränkt hast, damit sie den von sudo abhängigen pi Prozess killen. Danach beendet sich der sudo root Prozess auch.
 
> Aber nein, das wird nicht (mehr) benötigt. Als der legacy-run-server neu war, wurde atreboot noch als root ausgeführt. Da das nicht mehr der Fall ist könnte das sudo an der Stelle auch weg. Wenn es [...]

Da bin ich wieder anderer Meinung. Meiner Meinung nach schadet es, da es einen zusätzlichen sudo root Prozess benötigt. Eine entsprechende Dokumentation hast du doch schon vorbildlich oben drüber. 

> Der Grund warum ich das eingebaut habe ist weil ich dachte bash würde hier einen fortlaufenden File-Handle öffnen. In dem Fall würden sich mehrere Prozesse die gleichzeitig ins Log schreiben [...]

Ja, das wäre dann einfach simpel bzw. KISS. Der Konstrukt kann an beiden Stellen also atreboot.sh und LRS raus.

hhoefling, 2022-08-14:
Ich habe in atreboot.sh oben
```
SELF=`basename $0`
function log()
{
 timestamp=`date +"%Y-%m-%d %H:%M:%S"`
 echo $timestamp $SELF $*
}
```
drin, und dann alle (loggenden) "echo"s->"log" getauscht bzw noch ein paar weiter eingebaut.
Dazu haben die beiden Aufrufer von atreboot.sh weiterhin die Ausgabeumleitung zur Logdatei.
Ist noch simpler als openWBDebugLog zu benutzen und sieht auch anders aus.
Damit habe ich Nebenwirkunsfrei einen Datumstempel + Scriptname im Logging..
Ist zusätzlich leicht vom vom normalen (Regel)-logging zu unterscheiden.

DetMoerk, 2022-08-14:
@hhoefling: mach aus $timestamp noch "$timestamp:", dann wird es noch schöner :-)

hhoefling, 2022-08-14:
Wenn "%Y-%m-%d %H:%M:%S: "

DetMoerk, 2022-08-14:
Auch das kann man machen, aber in der Zwischenzeit würde ich auch noch hinter das $SELF einen : setzen ;-)

hhoefling, 2022-08-14:
Überlasse ich dir als Übung. :-)