snaptec/openWB #2358: verbraucher3

hhoefling, 2022-08-11:
Es gibt einige Variablen für verbraucher_3
Laut Gui gibt es aber nur 2 Verbrauchen ( for Schleife 1  bis <3 )
Weiter unten dann in PHP eine Schleife 1 bis <4 
Aber der restliche Code "versteht"  nur Verbrauche 1 und 2
Also kann alles mit "verbraucher_3*" entsorgt werden.

benderl, 2022-08-31:
Kannst Du die Stellen, an denen Du das bemerkt hast, angfeben? Datei und Zeilennummer wäre hilfreich.

hhoefling, 2022-08-31:
Da deine IDE anscheinend keinen Such(all) Befehl hat, hier mal das Ergebniss einer Suche auf
der 1.9.274 (stable)
```
E:\GitHub\snaptec_openWB\loadvars.sh [1]
  1060,19: if [ -s "ramdisk/verbraucher3_watt" ]; then verb3_w=$(<ramdisk/verbraucher3_watt); else verb3_w=0; fi
E:\GitHub\snaptec_openWB\runs\cron5min.sh [5]
  73,1: verbraucher3=$(<"$RAMDISKDIR/verbraucher3_wh")
  103,146: echo "$(date +%H%M),$bezug,$einspeisung,$pv,$ll1,$ll2,$ll3,$llg,$speicheri,$speichere,$verbraucher1,$verbrauchere1,$verbraucher2,$verbrauchere2,$verbraucher3,$ll4,$ll5,$ll6,$ll7,$ll8,$speichersoc,$soc,$soc1,$temp1,$temp2,$temp3,$d1,$d2,$d3,$d4,$d5,$d6,$d7,$d8,$d9,$d10,$temp4,$temp5,$temp6" >> "$dailyfile"
  219,3: verbraucher3dailyyield=$(echo "scale=2;($verbraucher3 - $i) / 1000" |bc)
  220,10: echo "$verbraucher3dailyyield" > "$RAMDISKDIR/daily_verbraucher3ikwh"
  292,392: hausdailyyield=$(echo "scale=2;$bezugdailyyield + $pvdailyyield - $lladailyyield + $sedailyyield - $sidailyyield - $einspeisungdailyyield - $d1dailyyield - $d2dailyyield - $d3dailyyield - $d4dailyyield - $d5dailyyield - $d6dailyyield - $d7dailyyield - $d8dailyyield - $d9dailyyield - $verbraucher1dailyyield + $verbrauchere1dailyyield - $verbraucher2dailyyield + $verbrauchere2dailyyield - $verbraucher3dailyyield" | bc)
E:\GitHub\snaptec_openWB\runs\csvcalc.py [1]
  85,32: # $ll6_____________ 14 17 $verbraucher3
E:\GitHub\snaptec_openWB\runs\initRamdisk.sh [4]
  351,30: echo 0 > $RamdiskPath/daily_verbraucher3ikwh
  360,24: echo 0 > $RamdiskPath/verbraucher3_watt
  361,24: echo 0 > $RamdiskPath/verbraucher3_wh
  362,24: echo 0 > $RamdiskPath/verbraucher3vorhanden
E:\GitHub\snaptec_openWB\runs\updateConfig.sh [10]
  1030,17: if ! grep -Fq "verbraucher3_name=" $ConfigFile; then
  1031,9: echo "verbraucher3_name=Name" >> $ConfigFile
  1033,17: if ! grep -Fq "verbraucher3_aktiv=" $ConfigFile; then
  1034,9: echo "verbraucher3_aktiv=0" >> $ConfigFile
  1036,17: if ! grep -Fq "verbraucher3_typ=" $ConfigFile; then
  1037,9: echo "verbraucher3_typ=http" >> $ConfigFile
  1039,17: if ! grep -Fq "verbraucher3_urlw=" $ConfigFile; then
  1040,9: echo "verbraucher3_urlw='http://url'" >> $ConfigFile
  1042,17: if ! grep -Fq "verbraucher3_urlh=" $ConfigFile; then
  1043,9: echo "verbraucher3_urlh='http://url'" >> $ConfigFile
E:\GitHub\snaptec_openWB\web\display\gauges\index.php [6]
  662,24: if(strpos($line, "verbraucher3_name=") !== false) {
  663,15: list(, $verbraucher3_nameold) = explode("=", $line);
  671,6: $verbraucher3vorhanden = file_get_contents('/var/www/html/openWB/ramdisk/verbraucher3vorhanden');
  674,6: $verbraucher3_nameold = trim(preg_replace('/\s+/', ' ', $verbraucher3_nameold));
  1189,9: var verbraucher3vorhanden = <?php echo $verbraucher3vorhanden ?>;
  1190,9: var verbraucher3name = "<?php echo $verbraucher3_nameold ?>";
```

benderl, 2022-08-31:
Das wird dann entfernt, wenn das alte SmartHome rausfliegt. Abgesehen von den "Web Hooks" kann der Bereich bereits komplett durch SmartHome 2.0 abgedeckt werden.

closed 2022-08-31 by benderl