snaptec/openWB #2362: Horizontal swipe gestures undermine kiosk mode (history)

webermax, 2022-08-14:
...additional CLI flags might help, e.g. (Chromium):

```
# home/pi/.config/lxsession/LXDE-pi/autostart

@chromium-browser --incognito --disable-pinch --overscroll-history-navigation=0 --kiosk <URL...>
```

benderl, 2022-08-15:
Can you please create a pull request?

webermax, 2022-08-15:
Love to, but so far I couldn't figure out how to address the internal display (unsuccessful keywords: "Kiosk, Chrome/Chromium, Firefox, Display...").

Is it a specific "module", are config files written directly?

webermax, 2022-08-15:
...this could be the correct location to fix it permanently(?)

https://github.com/snaptec/openWB/blob/master/web/tools/lxdeautostart

benderl, 2022-08-15:
Seems like there are two places to fix. The one you already found and another in `runs/atreboot.sh`. https://github.com/snaptec/openWB/blob/130dca32cfb5524e1fa816f2ff5eb9b4acab6e81/runs/atreboot.sh#L99
I would suggest that `atreboot.sh` also just copies the file you found.

Seems also a bit weired, as there is an update procedure to disable swige gestures: https://github.com/snaptec/openWB/blob/130dca32cfb5524e1fa816f2ff5eb9b4acab6e81/runs/atreboot.sh#L370

I will have a look at this and implement a file version check similar to the one used in 2.0. https://github.com/openWB/core/blob/dbb259a5dd048f22d0c106880eb7bf3956d90e75/runs/atreboot.sh#L80

webermax, 2022-08-15:
Hm, found it, too (commented in pull request).

So [atreboot.sh](https://github.com/snaptec/openWB/blob/130dca32cfb5524e1fa816f2ff5eb9b4acab6e81/runs/atreboot.sh#L370) searches for `@chromium-browser --incognito --disable-pinch --kiosk http://localhost/openWB/web/display.php` in [lxdeautostart](https://github.com/snaptec/openWB/blob/master/web/tools/lxdeautostart).

Then the script would replace it with the correct flag(!).

benderl, 2022-08-15:
Please check current master with #2364

closed 2022-08-15 by benderl