snaptec/openWB #2387: Falsche Anzeige von "Hausverbrauch"/"PV" Leistungswerten bei Solaredge WR/Speicher

marschallphilipp, 2022-08-31:
Beim Laden des Akkus wird diese Leistung nicht bei "PV" inkludiert, obwohl der Akku ja aus der PV-Anlage gespeist wird. "PV" zeigt dann nur die aktuell im Haus verbrauchte Leistung an. Der Hausverbrauch wird dann offensichtlich auch falsch berechnet (0 Watt...).

Wenn der Akku entladen wird, stimmt der Hausverbrauch ebenfalls nicht. Dann wird bei "PV" die Leistung angezeigt, welche aus dem Akku kommt und gerade ins Haus eingespeist wird, beim Hausverbrauch wird jedoch zusätzlich "Speicher-Entladung" mit einberechnet (also doppelt).
 
https://github.com/snaptec/openWB/blob/master/packages/modules/solaredge/device.py#L101
[image]

LKuemmel, 2022-09-05:
Hast Du in den Einstellungen des PV-Moduls "Batterieleistung von PV abziehen" aktiviert?

marschallphilipp, 2022-09-30:
Ich hatte es zwischenzeitlich mal getestet, es ändert allerdings bei mir scheinbar gar nichts.....

Ich vermute, es hat etwas damit zu tun, dass ich einen Hybrid Wechselrichter habe, welcher DC in den Speicher lädt und wieder entlädt und dann erst im Wechselrichter in AC Spannung wandelt!?

LKuemmel, 2022-09-30:
Diesen Fall sollte "Batterieleistung von PV abziehen" entsprechend berechnen.
Bitte den Debug Modus auf 2 stellen und einen kompletten Durchlauf der Regelung posten. Eventuell enthaltene Anmeldedaten bitte vorher unkenntlich machen.

marschallphilipp, 2022-10-03:
Was bedeutet ein "kompletter Durchlauf der Regelung"?
Und wo finde ich anschließend das Log File? Gibt es dazu eine Anleitung oder einen Forums-Beitrag?

LKuemmel, 2022-10-04:
Das Log ist unter Status->Debug Log zu finden. Dort den Teile zwischen zwei "**** Regulation loop start ****" kopieren.

marschallphilipp, 2022-10-08:
`2022-10-08 20:50:36: PID: 30140: **** Regulation loop needs 5 seconds (LV2) at 55 cleanup /var/www/html/openWB/regel.sh
2022-10-08 20:50:36: PID: 30140: Überschuss -23; mindestens 1500 (LV1) at 104 nurpvlademodus nurpv.sh
2022-10-08 20:50:36: PID: 30140: uberschuss -23 wattbezug 23 ladestatus 0 llsoll 0 pvwatt -513 mindestuberschussphasen 1500 wattkombiniert -23 schaltschwelle 230 (LV2) at 576 main /var/www/html/openWB/regel.sh
2022-10-08 20:50:36: PID: 30140: anzahlphasen 1 (LV1) at 575 main /var/www/html/openWB/regel.sh
2022-10-08 20:50:35: PID: 30140: Zeit zum abfragen aller Werte 3 Sekunden (LV1) at 176 main /var/www/html/openWB/regel.sh
2022-10-08 20:50:35: PID: 30140: chargestatlp1 0	chargestatlp2 0	chargestatlp3 0 (LV1) at 1305 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: plugstatlp1 0	plugstatlp2 0	plugstatlp3 0 (LV1) at 1304 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: lp1enabled 1	lp2enabled 1	lp3enabled 1 (LV1) at 1303 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: EVU 1:V/1A 2: V/-1A 3: V/-1A (LV1) at 1302 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: lla3 0	llv3 235.9	llas13  llas23  soclp1 54 soclp2 (LV1) at 1301 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: lla2 0	llv2 235.2	llas12  llas22  sofortll 16 hausverbrauch 1129 wattbezug 23 uberschuss -23 (LV1) at 1300 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: lla1 0	llv1 235.8	llas11  llas21  mindestuberschuss 1500 abschaltuberschuss 5 lademodus 2 (LV1) at 1299 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: pv1watt -513 pv2watt  pvwatt -513 ladeleistung 0 llalt 0 nachtladen 0 nachtladen 0 minimalA 16 maximalA 16 (LV1) at 1298 loadvars loadvars.sh
2022-10-08 20:50:35: PID: 30140: speicherleistung -593 speichersoc 79 (LV1) at 1293 loadvars loadvars.sh
2022-10-08 20:50:34: PID: 30332: EVU RET: 0 (LV2) at 16 main modules/bezug_solaredge/main.sh
2022-10-08 20:50:34: PID: 2535: legacy run server: Completed running command in 0.25s: ["bezug_solaredge.solaredge","192.168.2.53","502","1"]
2022-10-08 20:50:34: PID: 2535: SolarEdge EVU: Update completed successfully
2022-10-08 20:50:34: PID: 2535: soc.modules.common.store._api: Saving CounterState(voltages=[235.41, 234.92000000000002, 235.53], power_factors=[0.5659000000000001, -0.40090000000000003, -0.41600000000000004], power=23.0, currents=[1.8, -1.1, -1.0], frequency=50.0, powers=[236.0, -111.0, -101.0], imported=1549722, exported=14228043)
2022-10-08 20:50:34: PID: 2535: SolarEdge EVU: Beginning update
2022-10-08 20:50:34: PID: 2535: legacy run server: Received command ["bezug_solaredge.solaredge","192.168.2.53","502","1"]
2022-10-08 20:50:33: PID: 30286: BAT RET: 0 (LV2) at 20 main modules/speicher_solaredge/main.sh
2022-10-08 20:50:33: PID: 2535: legacy run server: Completed running command in 0.22s: ["speicher_solaredge.solaredge","192.168.2.53","0"]
2022-10-08 20:50:33: PID: 2535: SolarEdge Battery: Update completed successfully
2022-10-08 20:50:33: PID: 2535: soc.modules.common.store._api: Saving BatState(power=-593.0, exported=0, soc=79.0, imported=0)
2022-10-08 20:50:33: PID: 2535: SolarEdge Battery: Battery SoCs=[79.0], powers=[-593.0]
2022-10-08 20:50:33: PID: 2535: soc.modules.common.component_context: Update Komponente ['Solaredge']
2022-10-08 20:50:33: PID: 2535: SolarEdge Battery: Beginning update
2022-10-08 20:50:33: PID: 2535: legacy run server: Received command ["speicher_solaredge.solaredge","192.168.2.53","0"]
2022-10-08 20:50:33: PID: 2535: legacy run server: Completed running command in 0.19s: ["wr_solaredge.solaredgeall","192.168.2.53:502","1","none","none","none","0","0","0","0"]
2022-10-08 20:50:33: PID: 2535: SolarEdge: Update completed successfully
2022-10-08 20:50:33: PID: 2535: soc.modules.common.store._api: Saving InverterState(power=-513.0, currents=[0.91, 0.92, 0.92], counter=20992766.0)
2022-10-08 20:50:33: PID: 2535: SolarEdge: slave=1: power=5130*10^-1, energy=20992766*10^0, currents=[91, 92, 92] * 10^-2
2022-10-08 20:50:32: PID: 2535: SolarEdge: Beginning update
2022-10-08 20:50:32: PID: 2535: legacy run server: Received command ["wr_solaredge.solaredgeall","192.168.2.53:502","1","none","none","none","0","0","0","0"]`

marschallphilipp, 2022-10-08:
Scheinbar werden PV Leistung (da Hybridwechselrichter) + Speicherentladung + Netzbezug zusammengezählt für den Hausverbrauch. Das kann ja nicht stimmen....
pvwatt -513
speicherleistung -593
wattbezug 23
-------------------------
hausverbrauch 1129

benderl, 2022-10-08:
Hast Du inzwischen die aktuelle Nightly installiert? Es gab noch einige Anpassungen an dem Modul.

marschallphilipp, 2022-10-08:
Wenn ich "Batterieleistung von PV abziehen" auf "Ja" stelle ändert es sich nicht:

`2022-10-08 21:09:47: PID: 21825: **** Regulation loop needs 6 seconds (LV2) at 55 cleanup /var/www/html/openWB/regel.sh
2022-10-08 21:09:47: PID: 21825: LP1 Modbus 0 korrekt (LV1) at 236 evsemodbuscheck evsedintest.sh
2022-10-08 21:09:45: PID: 21825: Zeit zum abfragen aller Werte 4 Sekunden (LV1) at 176 main /var/www/html/openWB/regel.sh
2022-10-08 21:09:44: PID: 21825: chargestatlp1 0	chargestatlp2 0	chargestatlp3 0 (LV1) at 1305 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: plugstatlp1 0	plugstatlp2 0	plugstatlp3 0 (LV1) at 1304 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: lp1enabled 1	lp2enabled 1	lp3enabled 1 (LV1) at 1303 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: EVU 1:V/-1A 2: V/-3A 3: V/5A (LV1) at 1302 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: lla3 0	llv3 235.6	llas13  llas23  soclp1 54 soclp2 (LV1) at 1301 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: lla2 0	llv2 236.1	llas12  llas22  sofortll 16 hausverbrauch 4955 wattbezug 5 uberschuss -5 (LV1) at 1300 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: lla1 0	llv1 235.9	llas11  llas21  mindestuberschuss 1500 abschaltuberschuss 5 lademodus 3 (LV1) at 1299 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: pv1watt -2424 pv2watt  pvwatt -2424 ladeleistung 0 llalt 0 nachtladen 0 nachtladen 0 minimalA 16 maximalA 16 (LV1) at 1298 loadvars loadvars.sh
2022-10-08 21:09:44: PID: 21825: speicherleistung -2526 speichersoc 76 (LV1) at 1293 loadvars loadvars.sh
2022-10-08 21:09:43: PID: 22018: EVU RET: 0 (LV2) at 16 main modules/bezug_solaredge/main.sh
2022-10-08 21:09:43: PID: 2535: legacy run server: Completed running command in 0.31s: ["bezug_solaredge.solaredge","192.168.2.53","502","1"]
2022-10-08 21:09:43: PID: 2535: SolarEdge EVU: Update completed successfully
2022-10-08 21:09:43: PID: 2535: soc.modules.common.store._api: Saving CounterState(voltages=[235.69, 235.91, 235.33], power_factors=[-0.7689, -0.9909, 0.9929000000000001], power=5.0, currents=[-1.5, -3.3000000000000003, 5.0], frequency=49.95, powers=[-429.0, -763.0, 1197.0], imported=1549735, exported=14228049)
2022-10-08 21:09:43: PID: 2535: SolarEdge EVU: Beginning update
2022-10-08 21:09:43: PID: 2535: legacy run server: Received command ["bezug_solaredge.solaredge","192.168.2.53","502","1"]
2022-10-08 21:09:42: PID: 21971: BAT RET: 0 (LV2) at 20 main modules/speicher_solaredge/main.sh
2022-10-08 21:09:42: PID: 2535: legacy run server: Completed running command in 0.09s: ["speicher_solaredge.solaredge","192.168.2.53","0"]
2022-10-08 21:09:42: PID: 2535: SolarEdge Battery: Update completed successfully
2022-10-08 21:09:42: PID: 2535: soc.modules.common.store._api: Saving BatState(power=-2526.0, exported=0, soc=76.0, imported=0)
2022-10-08 21:09:42: PID: 2535: SolarEdge Battery: Battery SoCs=[76.0], powers=[-2526.0]
2022-10-08 21:09:42: PID: 2535: soc.modules.common.component_context: Update Komponente ['Solaredge']
2022-10-08 21:09:42: PID: 2535: SolarEdge Battery: Beginning update
2022-10-08 21:09:42: PID: 2535: legacy run server: Received command ["speicher_solaredge.solaredge","192.168.2.53","0"]
2022-10-08 21:09:42: PID: 2535: legacy run server: Completed running command in 0.11s: ["wr_solaredge.solaredgeall","192.168.2.53:502","1","none","none","none","0","0","0","1"]
2022-10-08 21:09:42: PID: 2535: SolarEdge: Update completed successfully
2022-10-08 21:09:42: PID: 2535: soc.modules.common.store._api: Saving InverterState(power=-2424.0, currents=[3.42, 3.43, 3.39], counter=20992990.0)
2022-10-08 21:09:42: PID: 2535: SolarEdge: slave=1: power=24240*10^-1, energy=20992990*10^0, currents=[342, 343, 339] * 10^-2
2022-10-08 21:09:42: PID: 2535: SolarEdge: Beginning update
2022-10-08 21:09:42: PID: 2535: legacy run server: Received command ["wr_solaredge.solaredgeall","192.168.2.53:502","1","none","none","none","0","0","0","1"]`

marschallphilipp, 2022-10-08:
> Hast Du inzwischen die aktuelle Nightly installiert? Es gab noch einige Anpassungen an dem Modul.

Nein, bisher nicht.

benderl, 2022-10-08:
Na dann mal los. 😄

marschallphilipp, 2022-10-12:
Ich bin nun auf Version 1.9.282,leider wird die Batterieentladung immer noch mit der PV Leistung zusammengezählt.

`2022-10-12 07:55:12: PID: 17997: **** Regulation loop start **** (LV1) at 77 main /var/www/html/openWB/regel.sh
2022-10-12 07:55:06: PID: 16434: **** Regulation loop needs 5 seconds (LV2) at 62 cleanup /var/www/html/openWB/regel.sh
2022-10-12 07:55:06: PID: 16434: Überschuss -22; mindestens 1500 (LV1) at 104 nurpvlademodus nurpv.sh
2022-10-12 07:55:06: PID: 16434: uberschuss -22 wattbezug 22 ladestatus 0 llsoll 0 pvwatt -284 mindestuberschussphasen 1500 wattkombiniert -22 schaltschwelle 230 (LV2) at 583 main /var/www/html/openWB/regel.sh
2022-10-12 07:55:06: PID: 16434: anzahlphasen 1 (LV1) at 582 main /var/www/html/openWB/regel.sh
2022-10-12 07:55:05: PID: 16434: Zeit zum abfragen aller Werte 3 Sekunden (LV1) at 183 main /var/www/html/openWB/regel.sh
2022-10-12 07:55:04: PID: 16434: chargestatlp1 0 chargestatlp2 0 chargestatlp3 0 (LV1) at 1300 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: plugstatlp1 0 plugstatlp2 0 plugstatlp3 0 (LV1) at 1299 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: lp1enabled 1 lp2enabled 1 lp3enabled 1 (LV1) at 1298 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: EVU 1:V/1A 2: V/1A 3: V/-1A (LV1) at 1297 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: lla3 0 llv3 232.7 llas13  llas23  soclp1 0 soclp2  (LV1) at 1296 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: lla2 0 llv2 232.1 llas12  llas22  sofortll 16 hausverbrauch 644 wattbezug 22 uberschuss -22 (LV1) at 1295 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: lla1 0 llv1 232.4 llas11  llas21  mindestuberschuss 1500 abschaltuberschuss 5 lademodus 2 (LV1) at 1294 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: pv1watt -284 pv2watt  pvwatt -284 ladeleistung 0 llalt 0 nachtladen 0 nachtladen 0 minimalA 16 maximalA 16 (LV1) at 1293 loadvars loadvars.sh
2022-10-12 07:55:04: PID: 16434: speicherleistung -338 speichersoc 27 (LV1) at 1288 loadvars loadvars.sh
modules.common.fault_state.FaultState: ('TCP-Client konnte keine Verbindung zu 192.168.2.53:502 aufbauen. Bitte Einstellungen (IP-Adresse, ..) und Hardware-Anschluss prüfen.', <FaultStateLevel.ERROR: 2>)
" aufbauen. Bitte Einstellungen (IP-Adresse, ..) und " + "Hardware-Anschluss prüfen.") from e
File "/var/www/html/openWB/packages/modules/common/modbus.py", line 93, in __read_registers
self.delegate.read_holding_registers, address, types, byteorder, wordorder, **kwargs
File "/var/www/html/openWB/packages/modules/common/modbus.py", line 119, in read_holding_registers
unit=component_config.configuration.modbus_id)) or 1
File "/var/www/html/openWB/packages/modules/solaredge/device.py", line 77, in add_component
id=num, configuration=SolaredgeCounterConfiguration(modbus_id=int(slave_id0))))
2022-10-12 07:55:04: PID: 16961: EVU RET: 0 (LV2) at 16 main modules/bezug_solaredge/main.sh
File "/var/www/html/openWB/packages/modules/solaredge/device.py", line 206, in read_legacy
parser.set_defaults(RUN=lambda args: function(*[getattr(args, argument_name) for argument_name in arg_spec.args]))
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 19, in <lambda>
args.RUN(args)
File "/var/www/html/openWB/packages/helpermodules/cli/_run_using_positional_cli_args.py", line 34, in run_using_positional_cli_args
run_using_positional_cli_args(read_legacy, argv)
File "/var/www/html/openWB/packages/modules/solaredge/device.py", line 272, in main
importlib.import_module(parsed[0]).main(parsed[1:])
File "/var/www/html/openWB/packages/legacy_run_server.py", line 109, in handle_message
self.__callback(read_all_bytes(connection))
File "/var/www/html/openWB/packages/legacy_run_server.py", line 87, in handle_connection
yield
File "/var/www/html/openWB/packages/legacy_run_server.py", line 48, in redirect_stdout_stderr_exceptions_to_log
Traceback (most recent call last):

The above exception was the direct cause of the following exception:

pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.2.53:502)]
raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
File "/usr/local/lib/python3.5/dist-packages/pymodbus/client/sync.py", line 107, in execute
return self.execute(request)
File "/usr/local/lib/python3.5/dist-packages/pymodbus/client/common.py", line 114, in read_holding_registers
address, number_of_addresses, **kwargs)
File "/var/www/html/openWB/packages/modules/common/modbus.py", line 83, in __read_registers
Traceback (most recent call last):
2022-10-12 07:55:04: PID: 2870: legacy run server: Unhandled exception
2022-10-12 07:55:04: PID: 2870: pymodbus.client.sync: Connection to (192.168.2.53, 502) failed: [Errno 111] Connection refused
2022-10-12 07:55:04: PID: 2870: modules.solaredge.device: Solaredge Bat-WR-gleiche IP: None, Externer WR: None, 2. Speicher: None, Speicherleistung subtrahieren: None 2. IP: None, Num: None
2022-10-12 07:55:04: PID: 2870: modules.solaredge.device: Solaredge Slave-IDs: [1, None, None, None]
2022-10-12 07:55:04: PID: 2870: modules.solaredge.device: Solaredge IP: 192.168.2.53:502
2022-10-12 07:55:04: PID: 2870: legacy run server: Received command ["modules.solaredge.device","counter","192.168.2.53","502","1"]
2022-10-12 07:55:04: PID: 16451: ##### cron5min.sh finished ##### (LV0) at 318 main /var/www/html/openWB/runs/cron5min.sh
[... cut ...]
2022-10-12 07:55:02: PID: 16451: electricity provider not set up; skipping price update (LV2) at 160 main /var/www/html/openWB/runs/cron5min.sh
2022-10-12 07:55:02: PID: 16451: Netzschutz konfiguriert; aktuelle Frequenz: 4994 (LV1) at 109 main /var/www/html/openWB/runs/cron5min.sh
2022-10-12 07:55:02: PID: 16451: daily csv updated: /var/www/html/openWB/web/logging/data/daily/20221012.csv (LV1) at 101 main /var/www/html/openWB/runs/cron5min.sh
2022-10-12 07:55:02: PID: 16451: ##### cron5min.sh started ##### (LV0) at 23 main /var/www/html/openWB/runs/cron5min.sh`

marschallphilipp, 2022-10-12:
Außerdem hängt nun "Laden" bei 3,46kW fest. (obwohl ich im Moment definitv nicht lade....)

[image]

benderl, 2022-10-12:
Bitte mal den Beitrag im Forum nachverfolgen. Dort werden die Zusammenhänge erläutert.
https://www.openwb.de/forum/viewtopic.php?f=9&t=5938

marschallphilipp, 2022-10-17:
> Bitte mal den Beitrag im Forum nachverfolgen. Dort werden die Zusammenhänge erläutert. https://www.openwb.de/forum/viewtopic.php?f=9&t=5938

Ich habe den Beitrag durchgelesen und dort auch noch einen Post mit meinem Anliegen erstellt.

benderl, 2022-10-17:
Dann schließe ich hier. Bitte im Forum weiter verfolgen und klären.

closed 2022-10-17 by benderl