snaptec/openWB #2400: CP Unterbrechung funktioniert nicht im "Nur Ladepunkt" Modus

andig, 2022-09-09:
Siehe https://openwb.de/forum/viewtopic.php?f=4&t=5646 und https://github.com/evcc-io/evcc/discussions/4010#discussioncomment-3596163

LKuemmel, 2022-09-09:
Die CP-Unterbrechung ist in #2294  enthalten.

andig, 2022-09-09:
@LKuemmel da steht:

> Phasenumschaltung und CP-Unterbrechung erfolgt in Threads und blockieren nicht mehr den Daemon.

bedeutet Deine Antwort zusätzlich, dass sie aktuell einfach gar nicht implementiert ist oder fehlt?

Kommt #2294 vor 2.0 oder müssen wir darauf warten? Wäre es- da der PR ja schon länger auf ist- möglich CP vorher nachzurüsten?

benderl, 2022-09-10:
Der PR und damit auch CP-U kommt rein, wenn die letzten bekannten Probleme mit den aktuellen Modulen behoben sind, da es doch eine größere Anpassung mit entsprechendem Fehlerpotential ist.

LKuemmel, 2022-12-09:
Die CP-Unterbrechung funktioniert mit der aktuellen Stable auch im NurLadepunkt-Modus.

closed 2022-12-09 by LKuemmel