snaptec/openWB #2404: Lademodus nach Start der openWB (in Einstellungen/Verschiedene) ist nicht persistent und fällt auf Sofort Laden zurück

mthunter1, 2022-09-13:
Die Einstellung / Verschiedene Einstellungen / Lademodus nach Start der openWB
ist nicht persistent. Nach dem Abspeichern springt sie zurück auf Sofort Laden.
Nach einem Reboot wird im Display "Lademodus" angezeigt.
Ziel ist es das es auch nach einem Reboot in PV startet (zumindest in meinem Fall)

https://openwb.de/forum/viewtopic.php?f=9&t=5833&p=71224#p71224

mthunter1, 2022-09-28:
Ich glaube ich bin heute mit Zufall über den Grund gestolpert.
Unter Einstellungen/System/Netzwerk Einstellungen ist bei PnP für virt.WLAN per default ein bootmode=3 definiert, der überschreibt offenbar jegliche Einstellung. Ich hab den Eintrag dort gelöscht und seither passt es oben.

closed 2022-10-07 by benderl

mthunter1, 2022-10-18:
@benderl  Bitte um Info was denn nun die implementierte Lösung ist. Wird der default Eintrag dort gelöscht und ein Hinweis hinterlegt, dass damit die StartUp-Einstellung nicht mehr funktioniert oder wurde der Fehler an sich wirklich korrigiert?
Danke

benderl, 2022-10-18:
Es gab keine Anpassungen an dem Code und das beschriebene Verhalten konnte nicht nachgestellt bzw. nachvollzogen werden. Heißt: das war ein nicht nachvollziebares "Einzelphänomen".