snaptec/openWB #2410: Ladeleistung falsch wiedergegeben

martingruening, 2022-09-19:
Setup: Zwei OpenWB series2 standard+, beide auf 1.9.277,
LP1 ("Carport") ist Master, LP2 ("Einfahrt") ist als "nur Ladepunkt" konfiguriert

Folgender Zustand ergibt sich (kein Fahrzeug zum Laden angeschlossen)

[image]

Problem: Es wird eine Ladeleistung angezeigt, die real gar nicht da ist (beide Ladepunkte nicht in Nutzung)

Nebenbei ist der Hausverbrauch auch noch falsch wiedergegeben mit 0 Watt (das ist aber nicht Gegenstand dieses Issues, nur eine Nebenbeobachtung). Alle Werte für EVU und PV kommen aus dem lokalen OpenEMS System.

Es zeigt sich das gleiche Bild auch in den MQTT-Kanälen:
openWB/global/WAllChargePoints = 1377
openWB/lp/1/W = 0
openWB/lp/2/W = 0

martingruening, 2022-09-19:
Nachtrag: im Debug Log finde ich häufig
`/var/www/html/openWB/modules/extopenwb/main.sh: Zeile 78: [: None: Ganzzahliger Ausdruck erwartet.`

Ich vermute einen Parsing Fehler. OpenEMS Version ist 22.4.

LKuemmel, 2022-09-19:
Der Fehler "Ganzzahliger Ausdruck erwartet" wurde in der Nightly beseitigt. Vor dem Update bitte ein Backup erstellen.

benderl, 2022-10-07:
Die aktuelle Stable enthält ebenfalls den Fix.

closed 2022-10-07 by benderl