snaptec/openWB #2472: Buchsenmotor nach Neustart vor Zerstörung schützen

Becker884, 2022-10-28:
Hallo,
im Forum und vom Support bekomme ich keine Antwort, daher probiere ich es hier.

Weiß Jemand wie bei einem Neustart sichergestellt wird, dass der GPIO 26 auf 0/low startet ?
In der boot/config.txt und in der /etc/rc.local ist nichts konfiguriert.
Erst mit der atreboot.sh gibt es eine Abschaltung.

Hintergrund der Frage ist: Mir ist nun zum 2. Mal der Buchsen Motor durchgebrannt und es gibt keine mehr.
Beim 1. Mal war ich selbst Schuld.
Beim 2. Mal suche ich noch nach meiner Schuld.

Habe nach dem 2. Durchbrennen den Buchsenmotorausgang der Addonkarte gemessen und festgestellt, dass dort beim Neustart für 30-60s 12V anliegen, **mit original Image**.

Darauf hin habe ich noch einen Kollegen gefragt, der hat 2x original Buchse, und nichts an der Software geändert und auch schon 2x den Motor kaputt gehabt. Er hat gemessen und meinte bei seiner Addon wäre 0V drauf.

Ich hatte ein frisches Buster Lite probiert und war so schlau in der config.txt `gpio=26=op,dl` einzufügen, dabei ist es aber PIN26 und nicht GPIO26.
Dennoch blieb der Motor beim 1. booten ruhig, daher dachte ich alles richtig gemacht und bin ins Haus gegangen zum installieren (und neustarten). Dabei schätze ich, ist der Motor irgendwann durchgebrannt, weil irgendwann der PIN26 auf 1/high gesetzt wurde.

Also muss es irgendeine versteckte Firmwarefunktion geben, die den Motor beim booten schützt, die sogar einen SD-Karten Wechsel zunächst übelebt.

MfG

benderl, 2022-11-25:
Es gibt keine "versteckten Firmwarefunktionen". Die Initialisierung der GPIOs wurde aber kürzlich noch überarbeitet.
Beim normalen Start (nicht Reboot!) sind einige Pins auf High, andere auf Low gesetzt. Dazu bitte die Chip-Dokumentationen lesen.

Becker884, 2022-11-26:
"GPIOs up to 8: default state is 1 (HIGH, or close to 3.3V)"
Der Buchsenmotor ist an GPIO7 - eigentlich ziemlich dumm, denn >9 sind default 0.
Was ist ein normaler Start ? Strom war aus, Strom an, bootet ?
Warum wird **nicht** in der config.txt der Buchsenmotor auf low gesetzt ?

benderl, 2022-11-28:
Das Design der AddOn-Platine und die gewählten GPIOs sind leider historisch so gewachsen. Im Nachhinein wäre es sinnvoller gewesen, für die Verriegelung der Buchse einen anderen Pin zu nutzen, der nach dem Boot automatisch auf PullDown eingestellt ist. Die Zuordnung läßt sich jedoch jetzt nicht mehr ändern.

Auf der AddOn-Platine ist ein zusätzlicher externer PullDown an den GPIO angeschlossen, der jedoch je nach Bauteiltoleranzen anscheinend im Grenzbereich liegt und ein Schalten des Aktors nicht zuverlässig verhindert. Weitere Details kann ich leider nicht nennen, da die Platine Closed Source ist.

Wir begegnen der Problematik jetzt in zweierlei Hinsicht:
1. Das Layout der Platine wird überarbeitet, damit auch beim Boot (GPIO auf Input, High) der Aktor zuverlässig nicht angesteuert wird.
2. In der `/boot/config.txt` werden alle von openWB verwendeten GPIOs so initialisiert, wie diese benötigt werden. Siehe #2532 

An dieser Stelle ein Danke für den Hinweis, dass diese Anpassung inzwischen möglich ist.

closed 2022-12-08 by benderl