snaptec/openWB #2480: Priorität mehrerer Geräte bei PV Überschuss

Dobby73, 2022-11-01:
Wie konfiguriere ich die Priorität mehrerer Geräte die nur bei Überschuss betrieben werden sollen?
Ich möchte einen PKW, eine Wärmepumpe und einen Heizstab per openWB so steuern, dass die Wärmepumpe immer Priorität hat, sofern über 3000W Überschuss vorhanden ist. Wenn nur 1000W - 2999W Überschuss da sind soll der Heizstab laufen. Und Wenn der PKW angestöpselt ist, soll er Vorrang vor dem Heizstab aber nicht vor der Wärmepumpe haben. 
Sind solche Szenarien mit openWB möglich?
Vielen Dank im Voraus für Eure Unterstützung :-)
VG Ralf

thesorrow101, 2022-11-02:
Hallo Ralf,

Bis auf das "Umschaltszenario" kann die openWB all das. Aber ich denke im Forum 
https://openwb.de/forum/index.php
bekommst du noch bessere Antworten.

LG,

Simon

benderl, 2022-11-02:
Gerade wenn externe Geräte mit einbezogen werden sollen, bitte im Forum melden. Anfragen zum SmartHome werden dort besser beantwortet.

closed 2022-11-02 by benderl