snaptec/openWB #2492: goecheck.sh Syntax Fehler

pungerer, 2022-11-12:
Im goecheck.sh sind mit version 1.9.**281** immer wieder folgende Einträge zu finden.

**goecheck.sh: Zeile 52: ((: >= 40: Syntax Fehler: Operator erwartet. (Fehlerverursachendes Zeichen ist \">= 40\").**

Firmware Version der Go-e ist 41

benderl, 2022-11-14:
Kannst Du das weiter debuggen?

benderl, 2022-11-15:
Die Aufrufe in dem Skript nachstellen und das erwartete Ergebnis abgleichen.

pungerer, 2022-11-15:
das hier konnte ich auf die schnelle finden:

[image]

benderl, 2022-11-17:
Laut der Fehlermeldung ist $fwv leer oder keine Zahl. Die wird anscheinend nicht richtig ermittelt.

pungerer, 2022-11-17:
wie im Screenshot zu sehen wird die fwv etwas unschön ausgegeben (041.0). Liegt hier das Problem?

pungerer, 2022-11-17:
Ich muss das System demnächst nochmal auf 1.9.281 patchen. Aber macht das hier wirklich sinn?

if (($fwv >= 40)) ; then
curl --silent --connect-timeout $goetimeoutlp1 -s http://$goeiplp1/mqtt?payload=**amx**=$current > /dev/null

Oben im Screenshot ist nur der parameter "amp" zu sehen aber nicht "amx"

xerion3800, 2022-11-17:
Bis Firmware 0.40 war es (glaube ich) nur möglich über "amp" die Ampere zu verstellen. Der Nachteil war aber dass das jedes mal in den Flash geschrieben wurde. Seit 0.41 kann man über "amx" die Ampere verstellen und das wird nicht in den Flash geschrieben. Das du beim Status aber nur "amp" siehst  ist normal, da dort der zuletzt geschriebene Wert   aus dem Flash steht. Dieser wird dann auch nach einem Neustart genutzt. Aber zur Regelung sollte man dann "amx" verwenden.

Ich habe auch zwei Go-e im Einsatz einen V2 und V3 konnte diese Einträge aber im Log nicht bei mir finden?

pungerer, 2022-11-17:
Danke für die Erklärung! Ich werde die Tage nochmal ein Update auf 1.9.281 wagen und das debug.log danach durchforsten.
Dann klingt das vorerst nach Entwarnung.

pungerer, 2022-12-04:
Ich habe nun ein update auf 1.9.288 (stable) gemacht und kann keine Logeinträge bezüglich goecheck.sh finden.

closed 2022-12-04 by pungerer