snaptec/openWB #2528: Ladestand der Tesla Powerwall 2 wird falsch angezeigt / Incorrect value of battery_level

WallE159, 2022-11-26:
Für den Ladestand (battery_level) der Tesla Powerwall 2 wird in openWB ein falscher Wert angezeigt, z. B. 9%, obwohldie Tesla-App 5% angibt.
Grund scheint die Notreserve in der Powerwall 2 zu sein, die für den battery_level nicht berücksichtigt wurde.

Problembeschreibung und Lösung unter https://github.com/home-assistant/core/issues/82297.

Der korrigierte Wert müsste demnach wie folgt lauten (der Term kann sicher noch vereinfacht werden):
battery_level = (100 * ((battery_level / 100.0) - (0.05))) / (0.95)

benderl, 2022-11-30:
Wie kann die "Notreserve" ermittelt werden? Hart auf 5% zu setzen macht keinen Sinn.

WallE159, 2022-11-30:
Ich bin leider noch nicht dahintergekommen, ob und wie die "Notreserve" ausgelesen werden kann, ein fixer Wert in der Berechnung macht natürlich keinen Sinn.
Ein (nicht ganz so schöner) Workaround wäre evtl., dass dieser Wert in der Modulkonfiguration Batteriespeicher angegeben werden kann, sobald Tesla Powerwall ausgewählt wird ... 
Müsste vom User immer wieder angepasst werden, sobald die Notreserve in der Tesla-App geändert wird. Aber wie häufig macht man das?

mkanet, 2023-11-17:
Hi, sorry I only know English.

I am currently using the below template sensor to get a more precise value.  
```yaml
template:
  - sensor:
      - name: Powerwall Charge Actual
        unique_id: powerwall_charge_actual
        state_class: measurement
        unit_of_measurement: '%'
        device_class: battery
        state: "{{ ((95 / 91) * (states('sensor.powerwall_charge') | float - 9)) + 5 | round }}"
```
I don't know what `sensor.powerwall_backup_reserve` does either.   The value stays at `24`.  I presume it is taken from the below:
[image]

However, the value in the Tesla app shows `20%` in above backup reserve .   I think this sensor might need to be adjusted too.