snaptec/openWB #2570: Min+PV Laden bis max. 100% ermöglichen

Alles-geregelt, 2023-01-01:
Hallo,
ich bin neu hier und hoffe, dass ich meinen Wunsch an der korrekten Stelle platziere - andernfalls bitte ich um Nachsicht...
Mein aktuelles Problem: 
Ich habe einen Kostal Plenticore Wechselrichter sowie einen BYD-Batteriespeicher. Wenn diie Batterie längere Zeit (ab ca. einem Tag) nicht an die Kapazitätsgrenze gestoßen ist (100% bzw. 5%), dann stimmt der Anzeigewert der aktuellen Batterieladung nicht mehr mit der realen Batterieladung der Hausbatterie überein und weicht zunehmend ab (10% oder auch mehr). Hintergrund ist wohl, dass die Batterieladung nur geschätzt werden kann, solange die Min.- bzw. Maxwerte nicht erreicht werden und die Schätzung halt relativ ungenau ist. Ich gehe davon aus, dass diese Problematik auch für andere Wechselrichter-Batteriekombinationen zutrifft.
Daher wäre es sehr hilfreich für mich, wenn als max. Startpunkt für Min+PV Laden statt 95% (aktueller Stand) auch 100% möglich wäre. Sobald eine Batterieladung von100% angezeigt wird, ist die Batterie nämlich wirklich voll. Die Anzeige 99% dagegen kann über mehrere Stunden konstant bleiben, bis sie irgendwann auf 100% umspringt - abhängig von dem Zeitraum seit der letzten Vollladung.
Diese Lösung wäre für mich wesentlich komfortabler und ich könnte unnötige Netzeinspeisung vermeiden, wenn ich mal den richtigen Zeitpunkt nicht erwische, um "Min+PV-Laden" (einmal täglich) manuell zu starten, nachdem die 100% erreicht sind. 
Aus meiner SIcht wäre es darüber hinaus sogar wünschenswert, den Startpunkt 100% als "Standardwert" zu nutzen, weil ich meine Hausbatterie abends mit einem möglichst hohen Wert (dann also 95%, für das Beenden von Min+PV-Laden) in die Nacht fahren lassen möchte. Sollte das aus mir unbekannten technischen Gründen problematisch sein, wäre auch eine Lösung "einmal Start bei 100%", anschließende Neustarts bei 95% hilfreich.
Ist diese Änderung bei Open-WB reallisierbar?
Allgemeiner Kommentar: ich finde OPEN-WB wirklich genial und wünsche allen Entwicklern ein frohes neues Jahr.

benderl, 2023-01-02:
Wenn ich Dich richtig verstanden habe, willst Du unter "Ladeeinstellungen" -> "PV & Min+PV" -> "Einstellungen für Min+PV-Laden" -> "Speicher-Beachtung" -> "Einschalt-SoC" 100% einstellen können?
[image]

Alles-geregelt, 2023-01-02:
Ja, damit wäre mein Problem gelöst!

PR #2572 (2023-01-03): allow up to 100% for speichersocminpv
fixes #2570
closed 2023-01-03 by benderl