snaptec/openWB #2600: Verschiedene Sungrow WR PV1/PV2

davipet1, 2023-02-05:
Hallo Leute,

ich habe einen Sungrow SH10 (Hybrid mit Speicher) und einen SG6 Wechselrichter an meiner PV Anlage.
Per Modbus angebunden bekomme ich aber nur den SH an PV1. An PV2 gebe ich "Sungrow", IP, Port und ID an.
Im Status ist folgender Fehler:

> modules.common.modbus <class 'modules.common.fault_state.FaultState'> ('modules.common.modbus Modbus Error: [Input/Output] [Errno 104] Connection reset by peer', <FaultStateLevel.ERROR: 2>)

IP, Port und ID passen. Im IOBroker z.B. bekomme ich die Modbus-Verbindung aufgebaut.

Woran mag das liegen? Habe jetzt das Problem, dass ein Teil der PV-Leistung gar nicht erkannt wird.

Gruß

David

davipet1, 2023-02-26:
Ich habe nun PV2 auf MQTT umgestellt.
Hier gibt es 2 Probleme:

Der Text unter PV2 sagt:

> Keine Konfiguration erforderlich.
Per MQTT zu schreiben:
openWB/set/pv/2/W PV-Erzeugungsleistung in Watt, int, positiv
openWB/set/pv/2/WhCounter Erzeugte Energie in Wh, float, nur positiv

Aber die Topics openWB/set/pv/2/W und openWB/set/pv/2/WhCounter existieren nicht!
Ist vielleicht openWB/pv/2/W und openWB/pv/2/WhCounter gemeint? Diese sind jedenfalls nicht readonly...

Weiter bleibt der Fehler im Stausfenster:

> modules.common.modbus <class 'modules.common.fault_state.FaultState'> ('modules.common.modbus Modbus Error: [Input/Output] [Errno 104] Connection reset by peer', <FaultStateLevel.ERROR: 2>)

Schade, so ist an weniger sonnigen Tagen die PV-Landung per openWB schlecht. Nutze noch selbst geschriebene Scripts.
Gruß

David

benderl, 2023-02-26:
> Aber die Topics openWB/set/pv/2/W und openWB/set/pv/2/WhCounter existieren nicht! Ist vielleicht openWB/pv/2/W und openWB/pv/2/WhCounter gemeint? Diese sind jedenfalls nicht readonly...

Nein, die Beschreibung passt schon so mit "set".

Die alte Fehlermeldung kannst Du ignorieren.

Maxify23, 2023-03-06:
Moin David 
Das Problem ist leicht komplexer. 
Es sind Aktuell zwei Probeleme bei der SG Auslesung vorhanden, 

Ich würde die erstmal gerne lösen und schauen, dass wir deinen SH auf Reihe kriegen, dann schauen wir das wir das zusammen bekommen. 

Der Reihe nach: 
-SH gemäß meiner Anleitung im anderen Fehlereintrag einrichten und Rückmeldung geben. 
-Auf Rückmeldung von mir hier warten bis eine neue OpenWB Version mit den fixes erscheint. 

VG Maxim