snaptec/openWB #2630: BYD am Sungrow WR - kein "Entladen"

davipet1, 2023-03-02:
Hi,

sorry wenn ich schon wieder einen Issue aufmache...
Ich habe festgestellt, dass bei meinem Batteriespeicher auch "Laden" angezeigt wird, wenn er entladen wird.
Es ist ein BYD am Sungrow Hybrid-Wechselrichter. Die Vorzeichen von Strom und Leistung der beiden Modbusregister wechseln leider auch nicht. Ich vermute da könnte es dran liegen. Es muss leider auch auf DC-Power vom WR geschaut werden ob laden oder entladen stattfindet.
Gruß

David

benderl, 2023-03-02:
Leider ist bei Sungrow derzeit der Wurm drin und damit meine ich nicht das Modul in openWB. Die Daten lassen sich nicht konsistent auslesen. Es gibt viele Unterschiede in Abhängigkeit davon, wie der Sungrow angebunden ist (LAN, WiNet Dongle LAN, WiNer Dongle WLAN) und mit jeder neuen Software Version ändert sich etwas.

Es gibt aktuell ein paar Beiträge dazu im Forum, die sich intensiv damit auseinandersetzen. Wir von openWB haben keine Möglichkeit, an verschiedenen Systemen zu testen, daher sind wir hier auf die Hilfe aus der Community angewiesen.

Sungrow will auch eine aktualisierte Dokumentation veröffentlichen, aber das wird noch zwei bis drei Monate dauern.

Also bitte im Forum mithelfen, damit wir das hoffentlich schneller hinbekommen.

davipet1, 2023-03-02:
Hallo benderl,
ich muss zu meinem Bedauern gestehen, dass ich das Forum noch nicht kannte.
Ich werde mich natürlich gerne dort an der Lösungsfindung beteiligen!
Gruß

David

benderl, 2023-03-02:
Kein Problem. Danke für das Verständnis.

Lutz

Maxify23, 2023-03-06:
Moin David 
Ich bin Maxim und befasse mit intensiv mit dem Sungrow. (Habe auch eine SH mit BYD HVM)
Der Fehler ist bekannt. Das liegt nicht am Code (OpenWB) sondern daran dass die Auslesung des WR nicht an allen Anschlüssen vernünftig läuft.

BItte stelle sicher dass: 
-OpenWB auf dem aktulessten Version hast 
-Dein WIInet Dongel und Wechselrichter auf dem aktuellsten Software Stand hast
-Am besten über die IP des Lan Anschlusses die OpenWB einbindest

 Es giebt drei Möglichkeiten den WR ins Heimnetz zu bringen: 
-LAN (hinter dem WII NET)  (funktioniert und getestet)
-WII Net WIFI (funktioniert und getestet)
-WII Net LAN  (GEHT NICHT auf FW Update warten)

Bei fragen gerne melden. 

VG Maxim

Maxify23, 2023-05-16:
Hat das FW Update und das Umstecken auf den LAN Port den Fehler behoben ?