snaptec/openWB #2635: Daten von SolarEdge WR werden falsch übernommen und angezeigt

deialsc, 2023-03-05:
Ausgangslage: SolarEdge Anlage 19,4 kW Peak mit 2 per Modbus verbundenen WR (ID 1 u 3) und mit SolarEdge/BYD 12 kWh Batteriespeicher und OpenWB mit Version 1.9.295 drauf.
Es hatte bis vor einiger Zeit recht gut funktioniert, aber seit letztem Update werden die Daten falsch aus dem WR ausgelesen. Wenn ich die Werte vergleiche die (etwa gleichzeitig) vom SolarEdge Portal und über die OpenWB angezeigt werden, dann sehe ich z.B. folgende Unterschiede:
                           OpenWB                          SolarEdge
PV:                        5,89 kW                         4,44 kW
Hausverbrauch:    2,11 kW                          0,62 kW
Speicher-Lad.:      3,82 kW                          3,98 kW

Speziell bei Hausverbrauch und PV stellt sich eine deutlich abweichende Größe dar, auch wenn man den zeitlichen Versatz berücksichtigt. Das Unangenehme dabei ist neben der falschen Darstellung auch die Tatsache dass der Ladevorgang bei PV-Laden  ggf. auch startet wenn nicht genug Strom von der PV kommt oder manchmal auch andersrum.
siehe Screenshot
[image]

Maxify23, 2023-03-06:
Was auffällt ist , dass bei der Berechnung:

Hausverbauch =
PV
(+ oder -) (PowerMeeter Wert Messwert (negativer Wert bei Einspeisung/Positiver Wert bei Bezug)) (+ oder -) (Batterieladeleistung (negativer Wert beim Aufladen/Positiver Wert beim Entladen ins Hausnetz))
- Ladeleistung

Der PV Wert:
PV (OpenWB) = PV(Solaredge) + (2* Hausverbrauch)
ist

OpenWB wird da glaube ich nicht viel machen können da es wohl einfach ein falscher Wert im PV Register ist. 

Gibt es ne Modbus Doku von Solaredge dazu?

deialsc, 2023-03-06:
Ja, da gibt's anscheinend was dazu, hab's mal hier drangehängt.

[sunspec-implementation-technical-note.pdf](https://github.com/snaptec/openWB/files/10900579/sunspec-implementation-technical-note.pdf)

Maxify23, 2023-05-16:
Gibt es bisher irgendwelche Änderungen am Fehler? Neues Update von Solaredge ?

deialsc, 2023-05-16:
Ich selber habe keine Infos dass irgendwas von irgendeiner Seite geändert wurde, aber mittlerweile werden die Werte wieder richtig angezeigt.

Maxify23, 2023-05-16:
Probleme die sich selber Lösen :D. 
Vieleicht war auch die Darstellung in der Cloud fehlerhaft und wurde Herstellerseitig gefixt. 

Dann kann das Issue ja von dir geschlossen werden. 

VG Maxim

closed 2023-05-16 by deialsc