snaptec/openWB #2747: Sungrow implementation not working

seaspotter, 2023-06-27:
Reference to Core software tree: https://github.com/openWB/core/issues/977

Since beginning of June Sungrow implementation is not working due to two untested and unverified commits in openWB core (2.0) and 1.9 nightly software.
Several tests were carried out without success of fixing this. Several remarks were made.

Bugfix for this issue is also pretty easy, delete this 2 commits and bring back the old implementation of version 1.9.300 stable then Sungrow will run again.

Commits: https://github.com/snaptec/openWB/commit/e4f753fa9195af1318a8fc550d78e0cfaadb2a84
https://github.com/snaptec/openWB/commit/21384e34d5785e24e98fcc506cfef8e00ee9e093

Reference to forum:

https://openwb.de/forum/viewtopic.php?p=87703#p87703
https://openwb.de/forum/viewtopic.php?p=87694#p87694
https://openwb.de/forum/viewtopic.php?p=87707#p87707

The author of the changes is also not responding on this issue and as already discussed here, he seems even not capable of proper testing of PRs https://openwb.de/forum/viewtopic.php?t=6920&start=20

benderl, 2023-06-28:
#2748

seaspotter, 2023-06-28:
Still not fully working same like in Core Software: https://openwb.de/forum/viewtopic.php?p=88006#p88006

EVU and PV Inverter data seems to be ok but battery not