snaptec/openWB #2778: Goe-charger FW 0.42 akzeptiert amx parameter nicht zum Steuern.

radi334, 2023-08-11:
Der Goe-Charger aktzeptiert in der FW version 0.42 wieder nur den amp Parameter um die Stromstärke zu steuern. 

im Goe-Modul goecheck.sh wird ab version >=0.40 aber der amx parameter verwendet. ab 0.42 muss es aber wieder der amp sein. 

Liebe Grüße

Christoph

Marv2190, 2023-09-17:
Tatsächlich funktioniert der amx auch, wenn vorher der AMP auf maximal gesetzt wurde!
Heißt steht der AMP auf 6A und man steuert mit amx 10A an, dann lädt man trotzdem nur mit 6A.
Da der GoE sich teilweise auf 6A zurücksetzt, funktioniert aktuell die Steuerung mit der OpenWB nicht mehr zuverlässig...

DasDepp, 2023-12-28:
Hallo,
ich habe das selbe Problem. Für mich liegt das Problem an den folgenden Code Zeilen in der set-current.sh:
`oldgoecurrent=$(echo "$output" | jq -r '.amp')
			if (( oldgoecurrent != current )) ; then
				if ((majorVersion >= 40)) ; then
					curl --silent --connect-timeout "$goetimeoutlp1" -s "http://$goeiplp1/mqtt?payload=amx=$current" > /dev/null
				else
					curl --silent --connect-timeout "$goetimeoutlp1" -s "http://$goeiplp1/mqtt?payload=amp=$current" > /dev/null
				fi
			fi`

Es wird ab einer Firmware >= 40 der amx Wert gesetzt. Zuvor wird aber der amp Wert angefragt und nur bei Ungleichheit erfolgt das setzen des amx Wertes. Nach meiner Interpretation sollte bei der Abfrage des Wertes ebenfalls die Firmware Version berücksichtigt werden und somit die Abfrage des amx oder amp Wertes erfolgen.

Marv2190, 2025-05-15:
Habe es nun extern gelöst. Immer wenn der Goe aktualisiert (MQTT), dann schaut ein Programm ob AMP = 32, wenn nicht wird auf 32 gesetzt. So gehts.

Leider hab ich weiterhin das Problem, dass die OpenWB zuverläßig unter 32A steuert, aber die 32A nicht mehr ansteuern kann (es bleibt der letzte Wert). Hab die OpenWB nun auf 31A begrenzt, damit geht dann alles (man muss nur auf die 32A verzichten)