snaptec/openWB #2796: Connection to Modbus TCP not working

atze09, 2023-10-25:
Hallo zusammen,

Mit der neuen Software 2.0 (genauer gesagt 2023-09-27 10:18:20 +0200 [380ac109e])
Bekomme ich weder mit einem Programm auf dem Windows Rechner noch mit EVCC Zugriff auf ModbusTCP der openwb.
Openwb Series2 is auf Nur Ladepunkt (secondary) eingestellt, muss ich sonst noch irgendwelche Einstellungen für ModbusTCP machen?

Ich würde gerne probieren, dass EVCC die Ladesteuerung über die openwb übernimmt.
Vielen Dank schon einmal

benderl, 2023-10-25:
Modbus wurde erst nach dem Release integriert. Du musst auf die aktuelle Master Version wechseln.

atze09, 2023-10-25:
> Modbus wurde erst nach dem Release integriert. Du musst auf die aktuelle Master Version wechseln.

Das war die Ursache, danke. Connection funktioniert, nur aktuell mit timeouts aber das ist ein anderes Thema.

closed 2023-10-25 by atze09