snaptec/openWB #2798: Ständige timeout Fehler bei EVCC Openwb 2.0

atze09, 2023-10-29:
Nach Umstellung von openwb 1.9 auf 2.0 bekomme ich nun ständig timeout Fehler bei EVCC und die Anbindung funktioniert nicht . An der Topologie wurde nichts verändert.
EVCC läuft auf Home Assistant und die openwb ist über modbustcp angebunden

Ein Ping ist unauffällig.

atze09, 2023-10-29:
Hier der ein Teil des logs 
2023-10-29 13:04:29,543 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc, Payload None darf nicht 'None' sein.
2023-10-29 13:04:29,547 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/10/get/soc_timestamp, Payload None darf nicht 'None' sein.

benderl, 2023-10-29:
Falsches Repo, wenn die Probleme mit software2 auftreten.

closed 2023-10-29 by benderl

atze09, 2023-10-29:
> Falsches Repo, wenn die Probleme mit software2 auftreten.

Danke, irgendwie finde ich aber das andere repo nicht, könntest du mir einen Link schicken? Danke

benderl, 2023-10-29:
https://github.com/openWB/core