snaptec/openWB #2803: [1.9.x] - MaxPriceForCharging (awattar) should use  "config / get" and "config / set" mqtt paths

ChristophCaina, 2023-11-21:
When you enable the Price based charging in your Wallbox, you can set the 'MaxPriceForCharging' from the Display.
The topic for this is: `openWB/global/awattar/MaxPriceForCharging`

The Topic should rather be use the /config/get (get the current value) and /config/set (set the current value) paths like:

`openWB/config/get/global/MaxPriceForCharging`
`openWB/config/set/global/MaxPriceForCharging`

This would make it easier to expose / use it externaly without using the Display, for example for integrations such as
https://github.com/a529987659852/openwbmqtt

benderl, 2023-12-11:
All topics are separated in to `openWB/...` and `openWB/set/...`.
Use the first one for reading and the second for publishing a new value.

Bucky2k, 2024-01-06:
Found this issue exactly for the same purpose: Feed a new max price threshold to the openWB. So I guess it's up to someone creating a pull request? Unfortunately I have no experience at all (incl. OpenWB ruleset) on how to extend and modify the MQTT structure

Bucky2k, 2024-01-06:
@ChristophCaina Enabling TLS for Mosquito on Home Assistant side seems to be an endeavor in itself. Can you confirm, that the openWB initiated bridge and "enable external control" will allow to modify the max charging price? Or can I save the effort, as it's not working anyway?

ChristophCaina, 2024-01-06:
Sorry, I can't confirm this.
I don't use a Bridge and my HomeAssistant is using the mqtt Broker of the Wallbox directly.

I haven't continued to try the above Information since I don't use this Feature by myself and didn't had the time to Work on the Integration

Bucky2k, 2024-01-06:
Thanks, but I am highly confused right now anyway... Somehow I found a way to set the MaxPriceForCharging, but didn't find the option before in the MQTT Exporer. This one works just using the always on MQTT server of the wallbox: openWB/set/awattar/MaxPriceForCharging

ChristophCaina, 2024-01-06:
ok - now I could have a look into the whole thing again.
The problem I do have (and why I initially created the topic) is, that the openWB integration is using the topic `openWB/global/awattar/MaxPriceForCharging` to get the current information / value.

the problem here is, that the SET Topic has a different structure / path.

For example, here's the code to Send a number to the OpenWB and get it:

```
 for description in NUMBERS_GLOBAL_COPY:
        description.mqttTopicCommand = f"{mqttRoot}/config/set/{str(description.mqttTopicChargeMode)}/{description.mqttTopicCommand}"
        description.mqttTopicCurrentValue = f"{mqttRoot}/config/get/{str(description.mqttTopicChargeMode)}/{description.mqttTopicCurrentValue}"
```

but it would require a complete change of this code to catch, if you want to set a new value for the awattar price...

I would need to read the `openWB/global/awattar/MaxPriceForCharging` to get the value
and would need to use the `openWB/set/awattar/MaxPriceForCharging` to write it...

while all other relevant changes would still use the same `/config/get` and `config/set`
structure.

I might be able to adapt this - but the idea was, that the openWB should use a 'standard' for everything ...

benderl, 2024-02-15:
This is in fact the standard for openWB. In `openBW/config/` all data for fixed configurations are processed. The "MaxProceForCharging" is a dynamic setting that is changed often. So it is not a more or less "fixed" configuration, but a dynamic setting.

closed 2024-02-15 by benderl