snaptec/openWB #2815: "openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche"

IAmKonni, 2024-02-02:
Hallo,

bei meiner aktuellen OpenWB Standard mit SW 2 (Branch master, aktuellster Stand (aber auch beim letzten Release-Stand)) bekomme ich im main.log die folgenden Fehlermeldungen:

`2024-02-02 17:35:33,960 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:36,653 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:36,697 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:39,405 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:39,436 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:42,160 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:42,237 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:44,924 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:45,003 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:47,666 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:47,742 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:50,427 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:50,511 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:53,162 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:53,200 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:55,900 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:55,961 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:58,638 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:35:58,703 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:36:01,414 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/chargepoint/3/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.
2024-02-02 17:36:01,479 - {helpermodules.setdata:339} - {ERROR:Setdata} - Payload ungültig: Topic openWB/set/internal_chargepoint/0/get/power_factors, Payload 2147450.88 liegt in keinem der angegebenen Wertebereiche.`

Ich betreibe die Wallbox im Secondary-Modus, der interne Ladepunkt hat bei mir laut UI die ID 3 und nicht 0 (warum auch immer). Vielleicht hat es damit was zu tun.

benderl, 2024-02-02:
Wenn es eine fertig gekaufte openWB ist, bitte einen Fehlerbericht aus den Einstellungen senden. Dann haben wir mehr Infos zur Verfügung.

IAmKonni, 2024-02-02:
Danke für den Tipp, habe ich erledigt und in der Problembeschreibung hierhin verlinkt.

closed 2024-02-04 by benderl