snaptec/openWB #2816: [Security] Command injection vulnerability

eldstal, 2024-02-03:
## Security advisory

A command injection vulnerability exists in OpenWB in all versions between Jan 29, 2021 and the current version (Feb 03, 2024). The vulnerability allows an unauthenticated user to execute arbitrary OS commands on the host, with full root privileges.

### Affected products
OpenWB starting with commit [bd0562d](https://github.com/snaptec/openWB/tree/bd0562d9ab0d14634d9f11db0b1a6f0fd2fcf395) (Jan 29, 2021) and up to the current version [5d2334a](https://github.com/snaptec/openWB/tree/5d2334a13606b098ca371af8c214057a7409d1de) (Feb 03, 2024).

### Steps to reproduce
1. Install apache2 and install openWB as indicated in the setup script
2. Perform a POST to `web/settings/saveconfig.php` as follows:
```
curl -vv -X POST --data "etprovideraktiv=1&etprovider=<noexist.txt;sudo reboot; #" http://192.168.52.32/openWB/web/settings/saveconfig.php
```
3. Observe that the system reboots

### Cause

`saveconfig.php` ([:145](https://github.com/snaptec/openWB/blob/dc8aa5e7700ed6c06f34a80afb943bc32982d96c/web/settings/saveconfig.php#L145-L145)) calls `exec()` with user input from `$_POST` included in the command line. By inserting semicolons, a malicious user is able to terminate the intended command and insert additional OS commands.

Similar vulnerabilities exist at `saveconfig.php:154` and `saveconfig.php:160`.

### Impact

Since the default installation adds `www-data` to sudoers with full privileges, this command injection vulnerability can be used to run commands with full root privileges.

### Proposed Mitigation

Use `escapeshellarg()` on any data under user control before including it in a command line such as `exec()` or `passthru()`.

benderl, 2024-02-05:
Thanks for reporting the issue. Please have a look at #2817. Does the PR close this vulnerability?

eldstal, 2024-02-05:
Thanks for taking care of this so quickly!

I've taken a look at the commit and it looks good to me.

Speedy1991, 2024-02-06:
**Can you please not post those stuff in a public visible scope?**

This is like a CVE10 - an attacker may obtain full root access to the machine.
Please take down the steps to reproduce and offer an email on the README to report security issues.

Also see [this](https://docs.github.com/en/code-security/security-advisories/guidance-on-reporting-and-writing-information-about-vulnerabilities/about-coordinated-disclosure-of-security-vulnerabilities#best-practices-for-vulnerability-reporters) github notes about security reporting.

I love my openWb but your project managment and your developement pipeline really needs some careful review.

PR #2817 (2024-02-08): fix command injection
fixes #2816
closed 2024-02-08 by benderl