snaptec/openWB #2850: Security Disclosure: Path Traversal to Remote Code Execution Vulnerability in `/web/settings/saveconfig.php`

Yeeb1, 2024-10-09:
The endpoint `/web/settings/saveconfig.php` processes POST requests in a way that does not adequately protect against path traversal, particularly in the context of dynamically constructing command lines for execution. The vulnerability arises from the fact that user inputs (`$_POST['etprovider']`, `$_POST['socmodul']`, and `$_POST['socmodul1']`) are directly embedded into `exec()` calls without sufficient validation to ensure that the input does not lead outside of the intended directories. 

[image]

While `escapeshellcmd()` is already employed to sanitize these inputs, it is primarily intended to escape shell metacharacters and does not prevent path traversal sequences such as `../../../../`. This allows for the creation of inputs that effectively bypass the intended use of `escapeshellcmd()` and manipulate the application to execute arbitrary commands or access files outside the web root directory.

[image]

Yeeb1, 2024-10-09:
I noticed that in previous disclosures (e.g., [#2816](https://github.com/snaptec/openWB/issues/2816)) there were discussions around responsible disclosure.

However, this repository still does not have a security policy set up ([guidance here](https://docs.github.com/en/code-security/getting-started/adding-a-security-policy-to-your-repository)). Additionally, the contact addresses in the `README.md` are now pointing to openWB 2.0, which is now maintained at this repository: [openWB/core](https://github.com/openWB/core).

For these reasons, I have refrained from disclosing all the details that allow for full code execution. If necessary, I can provide further details and the Python proof-of-concept (PoC).

benderl, 2024-10-09:
Thanks for the hint.
How would you solve this? Will a combination of `basename()`, `realpath()` and `file_exists()` be sufficient?
```php
if (file_exists($_SERVER['DOCUMENT_ROOT'] . "/openWB/modules/" . basename(realpath($_POST[user_content])))) {
    ... save to execute
}
```

PR #2851 (2024-10-11): check for valid modules
fixes #2850
closed 2024-10-11 by benderl